hzeller/rpi-rgb-led-matrix #1135: is their any function/method to stop and start the Matrix?

mushfiq-heraclius-xen, 2020-09-05:
@hzeller i',m implementing a simple server application that takes hex values/bytes from a client app, what i want to implement is very simple , i make a string from those byte/hex values and pass them to a function using c++ threads just as below:
code snippet
/* all variable name starts with a underscore are global variables 
void scrollTextonRGBLEDMatrix(RGBMatrix *canvas,FrameCanvas *offscreen_canvas,std::string textToScroll)
{
	int delay_speed_usec = 1000000;
	if (_speed > 0) 
	{
		delay_speed_usec = 1000000 / _speed / _font.CharacterWidth('W');
	} 
	else if (_x_origin == _x_default_start) 
	{
		// There would be no scrolling, so text would never appear. Move to front.
		_x_origin = 0;
	}

	int x = _x_origin;
	int y = _y_origin;
	int length = 0;

	while (_loops != 0) 
	{
		offscreen_canvas->Fill(_bg_color.r, _bg_color.g, _bg_color.b);
		// length = holds how many pixels our text takes up
		length = rgb_matrix::DrawText(offscreen_canvas, _font, x, y + _font.baseline(), _color, nullptr, textToScroll.c_str(), _letter_spacing);

		if (_speed > 0 && --x + length < 0) 
		{
			x = _x_origin;
			if (_loops > 0) 
				--_loops;
		}
		// Swap the offscreen_canvas with canvas on vsync, avoids flickering
		offscreen_canvas = canvas->SwapOnVSync(offscreen_canvas);
		usleep(delay_speed_usec);
	}	
}
int main(int argc, char *argv[])
{
     // codes that takes hex values from client using socket and then prepare the std::string text_to_scroll
    std::thread t1(scrollTextonRGBLEDMatrix, canvas, offscreen_canvas, text_to_scroll);
     t1.detach();
}
the problem with this code is whenever i pass a line of text  to thread t1 it scrolls perfectly, but my problem is i dont know how to clear the present scrolling text so whenever i pass a 2nd line, my RGB LED Matrix overlapped showing both first text and second text i passe as the previous thread is running as  daemon because of  using  t1.detach(), how can  i get rid of this problem?
TIA.

hzeller, 2020-09-05:
You do not want a start/stop matrix here. Your goal here is more related to how to deal with the thread.

If you want to have a text and replace it later, you should have exactly one thread, that you then pass a new text. Don't let loose of the thread, you need a reference to it to pass some data (ideally, you use a thread abstraction like the one provided in [this project](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/thread.h#L24) - that way, you have an object you can more easily abstract the passing data part).

Your goal is to run the thread and tell it when to show something different. You typically would do that by having a method on your thread `SetNewText(std::string s)` of sorts, which manipulates an instance variable. Inside the `Run()` method, you'd check with every loop if that text changed and do whatever is necessary to switch to the new display.

You **must** use a mutex for passing the thread boundary of course.

In the ThreadedCanvasManipulator you see an example how to pass some data safely with a mutex; in this case it is a running flag, but you'd do something similar
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/threaded-canvas-manipulator.h

daveythacher, 2020-09-06:
The thread is not a daemon it is orphaned, because you detached it. Really what you want is a one thread to draw the image for the display. You really should not be creating more than one of these threads. 

If you want you can throw new threads. However you will need to kill running thread first. Then you can create the new thread with the method and data you want. There are ways around this since this is not ideal.

The other way to do is to use mutex locks. Since this is threads you are sharing memory space, which allows data sharing without inter process communication. However you can still have hazard conditions.

You will need a main loop which holds the thread. There will need to be a variable which signals when a new string has been loaded so you can abort.

`// In thread:
// Global scope: std::string textToScrol
// Global scope: bool isNewString
// Setup canvas
// Creat offscreen_canvas
int delay_speed_usec = 1000000;
if (_speed > 0)
	delay_speed_usec = 1000000 / _speed / _font.CharacterWidth('W');
else if (_x_origin == _x_default_start)
	_x_origin = 0; // There would be no scrolling, so text would never appear. Move to front.
while(1) // Should enable shutdown logic if you plan to kill this an start another one. (Can simple kill if main thread is dying.)
{
	int x = _x_origin;
	int y = _y_origin;
	int length = 0;
	// reload _loops
	// getLock
	while (_loops != 0 && !isNewString)
	{
		offscreen_canvas->Fill(_bg_color.r, _bg_color.g, _bg_color.b);
		// length = holds how many pixels our text takes up
		length = rgb_matrix::DrawText(offscreen_canvas, _font, x, y + _font.baseline(), _color, nullptr, textToScroll.c_str(), _letter_spacing);
		if (_speed > 0 && --x + length < 0) 
		{
			x = _x_origin;
			if (_loops > 0) 
				--_loops;
		}
		scrollTextonRGBLEDMatrix(offscreen_canvas, textToScroll)
		// Swap the offscreen_canvas with canvas on vsync, avoids flickering
		offscreen_canvas = canvas->SwapOnVSync(offscreen_canvas);
		// releaseLock
		usleep(delay_speed_usec);  // This here will allow a new string to be added
		//getLock
	}
	isNewString = false;
	// releaseLock
}
// Do clean up if killing thread to load thread which does different operation. (OS will clean up if main thread dying.)

// Somewhere outside thread:
void setNewString(std::string newString)
{
	//getLock
	textToString = newString;
	isNewString = true;
	//releaseLock
}`

daveythacher, 2020-09-06:
You can use this structure to select a different operations also. Which will avoid the overhead of creating multiple threads.  

RGBMatrix original canvas should exist in the long running thread. If you create single thread it should be created and destroyed there. If you create multiple threads it should exist in thread that creates the background threads. It does not have to be the parent thread, it can be a child thread. However it should be thread which lives as long as the program lives or at least as long as the panel is required.

You could make this its own process, but I see that as pointless at this point. A lot of overhead for what you get. However there are cases where this can be useful.

daveythacher, 2020-09-06:
You can do inter process communication between threads like processes. Effectively what you are doing is attempting to decouple the timing of the two threads of execution. In the example I listed above the mutex is used to block both threads till the sync in state.

This is not required. You can use a queue, FIFO, pipe, socket, etc. to create some kind of memory structure which allows one to write and walk away. The other will then read it when it is ready to do so. This has some overhead with it, but for some applications the syncing overhead is far worse.

I do not know much about the context of this. So I do not know which is better suited. I also do not know much about stuff that may suggest forking or independent process being better suited than process.

mushfiq-heraclius-xen, 2020-09-06:
@hzeller @daveythacher i'm grateful to both of you :) , feeling blessed :)

closed 2020-09-06 by mushfiq-heraclius-xen