hzeller/rpi-rgb-led-matrix #1137: Request for info: What bus is in the 64x32 RGB LED Matrix - 4mm pitch panels from Adafruit (or in panels in general)?

Indifferent-Shiba, 2020-09-07:
I have been working with the 64 x 32 panels from Adafruit to make a Social Media Counter as a school project and found that spec wise I can't find much about them. I don't know if this is the place to ask, but does anyone know much about the kinds of computing buses they put int RGB LED panels?

hzeller, 2020-09-07:
The connection to these panels is called HUB75 ... but there is no clear spec, more a hodgepodge of things that are sortof compatible.

Mostly, it is a 6 bit wide shift register to clock in two rows of RGB on/off information, then control signals such as latching these, a selector which of the rows are affected and when to switch them on. So you constantly have to push data there to keep the pixels updated.

I wouldn't call this a 'bus' really. And there is not really much 'computing' going on on these panels either.

daveythacher, 2020-09-08:
The interface is called HUB75 most of the time, which is a 16 pin interface in 2x8 configuration. The data pins usually is the number of colors times the number of rows times multiplex ratio. Along with control signals like clock, latch, output enable, etc. There is a HUB75 with three address line, four address lines and five address lines. At least for the ones commonly sold by Adafruit and Sparkfun. Both have tutorials.

Pins: (Your panel)
6 Data lines
4 Address lines
1 CLK
1 LAT
1 OE
3 GND

This means you have 1:16 multiplex ratio. 32/16 = 2 sets of data. Each data set is 3 colors for RGB. Internally these are just six parallel shift register chains for the data. The shift registers are special in that the contain constant current generators. The address pins usually are just 4 to 16 demultiplexors which drive P channel MOSFETs for the rows. The shift registers drive the columns.

However some of the other panels like outdoor panels use higher multiplex ratios to enable higher brightness. Since the average current is reduced by lower multiplex ratios. This usually will increase the number of data pins and also increase the power consumption of the panel. However there are tricks to reduce the pin count using simple combinational/sequential logic. Other differences can exist, but for these what is above is likely to be what you have.

These are the shift register based panels using constant current LED drivers. Current set by resistor soldered to board which is white color corrected for 5V. However most panels will operate at say 3.3volts, but with less brightness and off balance colors. Other LED driver based panels also exist. These are not supported by this project and are not common from Sparkfun or Adafruit to my knowledge. 

The first type is PWM where instead of using soft-PWM across the SPI/serial bus to the shift register you send over whole 12-16 bits of data per column. You still multiplex with this type of panel. These usually use traditional PWM which is better for LED panels. This project uses BCM as PWM is very stressful to software. 

Never do hard real time, IO, high performance, etc. in software. Moderns times this is common but that will change. Software was never supposed to be done this way, but manufacturing has always been a problem. Software is breaking down against power and frequency, while manufacturing is moving parallel. So this will eventually work out to be correct.

PWM drivers can use a clock source internally to drive the PWM counters. However it is more common to see them make this external. Likely the panels will connect this to OE and it will be called the grayscale clock. This enables certain application logic to customize the behavior without additional internal logic.

The next type is an extension of PWM which is memory mapped (MM). This uses a local memory copy which automates multiplexing. However this is rarely done completely. This is likely due to cost and thermal considerations of the drivers. In summary each chip stores a section of the panels frame locally.

Like the PWM drivers a grayscale clock is used. However now when the counter overflows it automatically load the next multiplex row instead of resetting to the same value. External logic is generally used to drive the PMOS drivers, so whatever is clocking the grayscale clock must have some kind of counter or state information to support this.

PWM and MM drivers generally have control registers. Some support dynamic current gains for controlling the constant current. While shift registers generally do not. These drivers also support command sets via the data pins. Usually LAT and CLK. This can cause some state machine hell, but generally you can solve this by precomputing the values you load in. FPS is determined by the grayscale clock against PWM bit count.

The annoying part for software is the format of the 6bit data bus. It requires a lot of bit mangling which is computationally expensive. Then again so is bit banging. Basically you are forcing the computer to speak a suboptimal language. If you study computers you understand a few things. Software sucks a data/IO, but in reality everything is data/IO.

The issue here is the notion of software is actually incorrect. One could call a look up table software. A configuration file could be called software. However most classify software according to assembly. Most software is charges in capacitors. Since most of everything is about time and moving data around a lot of stuff basically works back to capacitance. Memory locality, DSP, silicon frequency vs area, architectural/computer organization, etc. 

>I wouldn't call this a 'bus' really. And there is not really much 'computing' going on on these panels either.

Computation is not real. Computation was once an occupation. It followed math and algorithms. The reality is there is computation everywhere, because computation is work. Engineering fundamentally looks at this whole thing differently than say Computer Science.

Through is there is two major classifications of "computing" which is control and processing. However processing is a subset of control in reality. Control is simple and honestly most of processing is made complex without a real point. There is really nothing hardware cannot do. In fact most hardware is better. 

[... cut ...]
Could software do what is in those panels. Perhaps, but would that be very efficient? Also what kind of resources would be required to support that? Would that be very efficient? How much power would be required? Overall I think there is a perspective issue.

These questions also kind of ask, what is software? What makes it "tick"? Hardware is your friend, look down on it all you want. It could have made your life so much easier. It would cost more, but that is an argument with a lot of counters.

daveythacher, 2020-09-08:
These are generally very similar there is a number of different parts used. Likely those panels will work with this project. You can drive up to three panel chains in parallel with this project. It will consume most of your Pi's pins. This project will work with both of the Adafruit Pi adapters.

Things to be aware of is these generally have 25MHz data limits. However others have reported success above this, but I do not recommend this. Note this is at 5volts. Also this project bit bangs and does not have a completely known performance expectation. This means while you can drive a decent number of panels you may not get full performance possible. Most noticeable in the PWM bit counts.

Indifferent-Shiba, 2020-09-11:
Thank you for your guys' insights! I think I understand the panels better now, and realized that I was looking at them at the wrong angle.

closed 2020-09-11 by Indifferent-Shiba