hzeller/rpi-rgb-led-matrix #1138: 128x64 Single Panel questions

CuriousJames, 2020-09-07:
Hello, I bought this 128x63 px panel: https://www.aliexpress.com/item/4000029775879.html?spm=a2g0s.9042311.0.0.ccd14c4dqOTCrY

I see there is a special pin for 64x64, I assume I'll need to use that, but is there another pin for 128?  Or is this more than can be handled with this library?

Lastly are you able to tell me the connector name for the data connection to these pannels please so that I can look to buy one for any PCB design I might make in the future.  I'm not sure if it would have a special name on account of the notch, never ordered a notched PCB header before.

Thanks for making this library btw!

CuriousJames, 2020-09-07:
Found the connector name btw 16way IDC header

CuriousJames, 2020-09-07:
Also I noticed it says 'HUB-75E' instead of the 'HUB-75' I've seen on images on this project, I'm guessing the 'E' is just indicative of that extra pin? As opposed to it being a newer (enhanced) protocol?

Thanks!

daveythacher, 2020-09-08:
HUB-75E usually denotes 5 address line model. So it has address pins: A, B, C. D and E. 

Which also means you have a 1:32 multiplex ratio. Since you have 64 rows this means you have 2 rows active at anytime on the panel. Since this is RGB or three colors, then you will have 6 data pins. Two sets of three data pins. Along with that there is likely a clock, latch and output enable pin. Internally there is 6 shift register chains which are 128 bits in length each.

https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet

CuriousJames, 2020-09-08:
So it is possible to control this with your library?  Fantastic!  I'll recheck where it says what RPI I should use for chain 1 E pin as I didn't catch that, and did you say it would have a second clock?  Does that need to go to the same pin as the first clock on the pi?

The seller is going to send me a pinout soon so I'll be able to get this going, my smaller one already works really well!

James.

daveythacher, 2020-09-08:
This is not my library. I have not written a single line. I disagree with the approach(s) used here on many levels. I have no intention of ever committing to it. However I am semi-familiar with aspects of it. I also have looked into LED matrix panels and have a fair amount of knowledge of aspects of them also.

Are you connecting this with the Adafruit hat? If so you will be using a pin mapping which will only use one chain. There is a pin mapping file which you can use as template to wire it or define your own mapping. 

Here is section of it for the adafruit boards: https://github.com/hzeller/rpi-rgb-led-matrix/blob/55fa32fc2e02afb254ac834aea93589d5b891a11/lib/hardware-mapping.c#L70-L115

It would need a modification for the E pin. See: https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet
It would also need a modification for the PWM version (recommended). See: https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker

If using standard adafruit version run with --led-gpio-mapping=adafruit-hat.
If using PWM version run with --led-gpio-mapping=adafruit-hat-pwm.

To my knowledge there is only one clock on panels supported by this project. I am not sure what you are referencing.

daveythacher, 2020-09-08:
You lose Pin 4 with the PWM trick. This really should be mentioned in the readme, but it is not. 

You may be able to keep it with a different modification approach, but I am not sure how experienced you are with this. 

There is an issue of accidentally driving the pins to different values causing a high current flow which damages the pin drivers. Which could be permanent, however maybe the Pi prevents. (I have not tried it myself.) This assumes both are set to output drivers. 

This mistake can happen a number of ways. Since user space programs can drive the pins and reconfigure them. If a library or kernel module is loaded without being verified there can be a conflict. The Pi is a large code base and allows flexible coding, the expectation is you have full ownership of it. You do not distribute ownership without close inspection. 

This library attempts to work around this with the mappings. This is not free, but a nice feature.

Generally you attempt to lock it down with privileges or single context. However more than likely the privileges are being bypassed. This a general purpose operating system so there is no single context here. RTOS and controllers generally make the firmware single context, use tasks for specific sections which split resources directly, or maybe require CPU kernel mode execution to change pins. 

The HUB75 is normally safe since it drives all inputs. This kind of issue only really happens when output tries to drive against another output and they drive different values.

One of the issues with bit banging. Besides possible damage it can cause random behavior on panel.

daveythacher, 2020-09-08:
The Pi only has two PWM resources which some version use for sound. You can switch over to i2s or USB audio if you need it. However you will need to verify pinouts. Here is link for that: https://github.com/hzeller/rpi-rgb-led-matrix#bad-interaction-with-sound

The library can work without the PWM but will not be as stable. Especially if fully loaded which is not recommended anyhow.

Also here is the privileges stuff: https://github.com/hzeller/rpi-rgb-led-matrix#running-as-root

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin