hzeller/rpi-rgb-led-matrix #1142: Adafruit 64x32 Matrix shows nothing

StealthyNiffler, 2020-09-09:
I've recently bought a 64x32 P4 LED-matrix from Adafruit and tried to connect it to my raspberry pi zero WH according to your wiring-diagramm (I only made the connections marked with [1] and 😄 and because I have 32 rows I also connected D). However running any of the demos (with --led-rows=32 --led-cols=64) has no effect on my matrix. I've already deactivated the sound in my /boot/config.txt and I also blacklisted this snd_bcm2835 module mentioned in the Troubleshooting section, but these changes had no effect.
Thanks for your help in advance!

daveythacher, 2020-09-10:
Check power supply. Make sure you run as root. Try increasing gpio-slowdown. Increase led-pwm-lsb-nanoseconds. What chips are on the Panel?

daveythacher, 2020-09-10:
Also reduce the number of led-pwm-bits.

~/rpi-rgb-led-matrix/examples-api-use/demo --led-rows=32 --led-cols=64 --led-chain=1 --led-parallel=1 --led-show-refresh --led-slowdown-gpio=2 --led-pwm-bits=3 --led-pwm-lsb-nanoseconds=300 -D4

StealthyNiffler, 2020-09-10:
I've tried this but the matrix is still showing nothing. I am using a 5V 4A power supply. 
I started thinking that the board might be defect. Is there an easy way of testing if it is even capable of showing something?

Here are two pictures of the back of the board:
[image]
[image]

daveythacher, 2020-09-10:
That power supply is no where near strong enough to handle this panel. I am thinking you have 1:16 32x64 RGB panel. Which means that at 5 volts: 20 * 32/16 * 3 * 64 = 7680/1000 = 7.68A Now you may be pulling less than 20 per LED on average. 

However the way this panel works is with BCM. That puts the LEDs to full brightness briefly, so unless you use a lot of capacitors you will brownout that power supply. Likely a safe circuit is activating within the power supply.

Lets try a different demo which should use less power:
~/rpi-rgb-led-matrix/examples-api-use/clock -f ../fonts/7x13.bdf -d "%A" -d "%H:%M:%S" --led-rows=32 --led-cols=64 --led-chain=1 --led-parallel=1 --led-slowdown-gpio=2 --led-pwm-bits=3 --led-pwm-lsb-nanoseconds=300

Do you have a voltmeter? Does it support continuity? 

I was able to look up the parts:
https://www.mikrocontroller.net/attachment/392914/ICN2037.pdf
http://www.ruichips.com/uploads/file/20180815/20180815094021000.pdf

StealthyNiffler, 2020-09-10:
The matrix still shoes nothing :(. Unfortunately I do not own a voltmeter.

daveythacher, 2020-09-10:
Have Jumper wires, LED and resistor?

StealthyNiffler, 2020-09-10:
No all the electrical components I own, I habe specifically ordered for this project so I habe pretty much only what can be seen in the pictures.

daveythacher, 2020-09-10:
Can I see your power supply? Can you get a voltmeter?

daveythacher, 2020-09-10:
I think I know the power supply you have.

Just out of curiosity do you know how to code? Python, C, C++, etc. One thing you could do is test the shift registers function. Its kind of on the long shot side. It could prove some level of function. We can prove the power supply is applying power to the panel. We can also prove the shift registers are operational.

If the register is powered and operational it should work. We need a pattern to avoid random glitches and that the shift registers are driving the pins. The pattern could be a sequence like counting in hex on nibble boundaries. (Not the best pattern.) We can drive the OE high for this to limit the power consumption.

Drive pin R1 as the data on the input connector with jumper. Drive the CLK pin on the pin connector. Then look at the output connectors R1 value. It should match after 64 pulses. We can ignore G1, B1, R2, G2, B2, A, B, C, D, and LAT. OE should be driving high to disable the LEDs.

A simply bit banging programing should work. We will want to slow the pin access to make sure it does happen too fast.

If does not work we still have unknowns like power supply, board functional status, wiring, etc. If it does work we could repeat the test for G1, B1, R2, G2, and B2. We could sending OE low after driving one row. LAT, A, B, C and D are hard to test without voltmeter.

Me personally I would probe it out.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828