hzeller/rpi-rgb-led-matrix #1143: CPU usage and reserve core related

samuelbles07, 2020-09-10:
Hi,

Thank you for creating this amazing library, once again thank you!

Currently i'm tinkering about doing other heavily cpu usage (not always running, like converting mp4 video to .stream) while `led-image-viewer` run .stream is still running. I notice that in `utils` folder example, section `video-viewer` says 

> create a preprocessed stream first, then replay it with led-image-viewer. This results in best quality (no CPU use at play-time) ...

and in main `README` section CPU use,  we can reserve one core just for the refresh display by isuing `isocpus=3` in `/boot/cmdline.txt`. I've done it.

**Test 1, run .stream only**
When i run `led-image-viewer` .stream , and check cpu usage using `htop`, i saw that only core 4 usage around 80% and others are 1% / 2% / 3% with lowest refresh rate 220hz.  see pic below

[image]

**Test 2, run converting .stream to mp4**
I only see 1 core, which is core 2 cpu usage at 100% and core 4 not used at all. see pic below

[image]

**Test 3, simultaneously converting and run .stream**

See result video here [https://drive.google.com/file/d/1YRzCJTIii4gBsiR4bA2k24_9q8cTGwkB/view?usp=sharing](https://drive.google.com/file/d/1YRzCJTIii4gBsiR4bA2k24_9q8cTGwkB/view?usp=sharing)

As you can see, refresh rate of the display is dropping when converter start, and i also see that core 4 cpu usage increased by 4-5%.

**Question**

Why refresh rate is dropping when i already reserve core 4 only for display refresh rate and as result **Test 2**, only core 2 is used when converting. I don't understand what happened. Can anybody explain?

Thank you!!

daveythacher, 2020-09-10:
Best as I can tell the reason why nothing runs on core 4 is because updater thread never starts. When you define the output stream option it blocks it from ever starting. There is two roles defined in the logic and based on the options you give, one role becomes active and the other dormant.

Test 1 shows off the stream to display function. Test 2 shows off the convert to stream function. Test 3 shows the flaws in this project's design.

What happens under the covers is a lot of bit banging. Basically the CPU does everything, and it is critical that it is not interrupted because of this. So what the project does is attempts to block anything from getting in its way. Which is why the updater thread is assigned to its own core. However resources are shared between the core. The IO space, RAM, L2 cache, etc.

The operating system can decide to schedule what ever it deems fit, which is solved with the isolation trick. However if one of those tasks decides to evict a lot from the cache or eat memory bandwidth then problems can emerge. Also if multiple cores start writing to the IO space then there can be arbitration issues.

In summary this creates a slippage in time or access, which is visible in lower performance/quality.

Bit banging is not a good idea in microcontrollers and it is a lot easier to pull it off there. Single context, no operating system, generally a single, Harvard architecture, custom hardware, better DMA, timers, interrupts, etc. No you can solve almost all of that, and can make something work which is not based around this. That is a large topic.

In user space it is kind of a violation of user space vs kernel space policy of a general purpose operating system. It breaks the notion of processes, privileges, users, etc. In addition to all the other things.

One way to avoid this is to build a server and make the Pi a client over USB or Ethernet. Have the server do the heavy work of converting then pass that data to a light client application which update the matrix using offscreen canvas. Then swap the frames.

There is a project which did this already, but I would have to look for it. I think it is mentioned somewhere.

samuelbles07, 2020-09-11:
Hi @daveythacher , thank you for your great explanation.

My goal is, only using 1 raspberry pi for display and converting that sometimes simultaneously.  So based on your explanation, it is not possible to do both in 1 raspberry pi. 

I already did client-server design as you mention, but maybe from the project that you've seen more efficient, can you send the link here if it's okay?

Thank you!

daveythacher, 2020-09-11:
> Hi @daveythacher , thank you for your great explanation.
>
> My goal is, only using 1 raspberry pi for display and converting that sometimes simultaneously. So based on your explanation, it is not [...]

It technically is. If you use MM LED Panel, microcontroller and completely different library. (Which I am not aware of.) This would be low cost, easy to build, and fairly flexible. It would still some issues but not nearly as much as this one does. It would also be a little less than ideal, but I think to a degree a lot more defendable.

daveythacher, 2020-09-11:
Here is the link: https://github.com/hzeller/flaschen-taschen

This project mentions it here: https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#video-viewer

samuelbles07, 2020-09-15:
Thank you for the explanation dave, guess i close this issue.

closed 2020-09-15 by samuelbles07