hzeller/rpi-rgb-led-matrix #1144: Pin mapping a CM3+ for 6 parallels

KentWalker, 2020-09-10:
Hi Henner,

This isn't so much an issue but I have questions and didn't know where to ask.
Are there any particular GPIO pins that are more suited to tasks (eg PWM) than others on the CM3+ module?  I am updating an interface board to 6 parallels.  I am only using outdoor displays that are 32 x 16, 1/2 or 1/4 scan so only need the ABC lines.
Is there a particular mapping that is code efficient or doesn't it matter? I need to keep GPIO14 and GPIO15 free for comms.

Thanks for an awesome library.

hzeller, 2020-09-10:
I am not aware that  CM3 is too much different from the rest. The only important ones are PWM, as they are tied to hardware internally, and I think GPIO 0 and 1 is reserved by the Raspberry Pi people for HAT identification communication; but I think they should just work.

I have personally not tested the CM3+. It was the work by @arahasya  and I have not yet wired up panels yet myself (maybe I get a prototype board).

KentWalker, 2020-09-10:
Thanks Henner.  Which line(s) require PWM, is it the OE line?

hzeller, 2020-09-10:
Yes - that line requires precisely timed pulses, for which the PWM hardware is, let's say, repurposed.
GPIO18 and GPIO12 are capable of providing that.

KentWalker, 2020-09-10:
Ok, I'll stick with GPIO18 for OE.  Is there any benefit in grouping any lines, eg all the RGB lines in sequence for each parallel?  Are the GPIO pins accessed in byte/word/dword blocks or are they individually controlled bit-wise?  If so I will try and arrange the mapping to suit otherwise it will be whatever is easiest to route.  Thanks

hzeller, 2020-09-11:
The GPIOs are bundled together as one 64 bit word (in the case of CM3) and then sent, so don't worry about grouping them in particular ways.

KentWalker, 2020-09-11:
Great. Thanks for help

closed 2020-09-11 by KentWalker

arahasya, 2020-09-11:
If you are using GPIO 0 and GPIO 1 for output
You must add
force_eeprom_read=0
in /boot/config.txt

arahasya, 2020-09-11:
also my suggestion is to use GPIO36 and GPIO37 with ALT2 instead of 14 and 15.