hzeller/rpi-rgb-led-matrix #1145: Flickering on Raspberry pi 4 and 128x64 matrix

rijotech, 2020-09-10:
Hi, I'm facing some flickering issues with a raspberry pi 4  8 panels k716-128x64-32s and active-3 board
I used --led-slowdown-gpio=5 but still there is some fikering and I was wondwering how to fix it?

https://www.youtube.com/watch?v=pABox-gQTrE

sudo video-viewer -f -F demo.mp4 --led-rows=64 --led-cols=128 --led-chain=4 --led-parallel=2 --led-pixel-mapper=V-mapper --led-slowdown-gpio=5 --led-brightness=90

hzeller, 2020-09-10:
Short data and power cables are important. And yes, more slowdown can help as well.

It looks like the chips have glitches, which could be due to power stability (given that you already slowed down).
Make sure to have stable power, short, thick wiring from the power supply to the power inputs of the panels and possibly a low ESR cap directly at the power input. And possibly run an extra ground line to the Pi.

The digital chips are powered from the same source that creates huge current surges for the LED driving, and that might make their logic volatile, depending on how well the bypass capacitors are designed in on the matrix board. Adding some Low ESR capacity directly at the power input connector is probably a good experiment.

rijotech, 2020-09-11:
I see like what you did here https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power
Let me try.

daveythacher, 2020-09-11:
Try --led-pwm-bits=6. This uses BCM over PWM, so the current is very spiky.

You can try to work this out with caps, but there are other things here worthy mentioning. You are using video-viewer, which could stress processor. The display is ran from bit bang and if the CPU is loaded down this can reduce performance of display. 

You currently have the PWM dialed up which will increase the time per frame. This can reduce performance. It can also decrease the spike width for the lower bits. Lowering the PWM bits will remove these.

You are using the recommended OE time which will cause the spikes to be 130ns for the least significant bit. You can increase this but this will reduce frame rate some. However it could help with other things also, maybe. (Probably at the expense of this.)

daveythacher, 2020-09-11:
User settings:
pwm_bits = led-pwm-bits
oe = led-pwm-lsb-nanoseconds
columns = --columns

Panel settings:
multiplex_ratio

oe = (2^(11-pwm_bits))*(columns)(f / 1,000,000,000)
row_time = ((2^(11)) - (2^(11-pwm_bits)))*oe
frame_time = (multiplex_ratio * row_time)
refresh_rate = 1 / frame_time

Now f is the wild card. If the system is loaded it will be less. If you use a lot of GPIO slowdown, that is not required, f will be lowered.

I think you can pick 1-25 MHz for f. Pick too low and you get a higher than expected refresh rate. Pick too high and you get a lower one.

The good news is you should be able to find f with a few trials. (The program could does this automatically.)

Related: #981 #969 #971

daveythacher, 2020-09-11:
Note row_time is defect. However that is how it works. oe and pwm_bits can be set independently. The equations above suggest how to configure it to make it work correctly. I have seen people think they are getting better performance they really are.

There is a few advantages to do it correctly, but it can also cripple performance some also.

rijotech, 2020-09-11:
The capacitors did the trick, once again THANKS A LOT Mr. Henner Zeller for sharing your knowledge with the wolrd!

closed 2020-09-11 by rijotech

daveythacher, 2020-09-11:
@rijotech Becareful with that cap trick, there is a lot that he did not do. Which I will not mention.

hzeller, 2020-09-11:
David, I'd like to ask you again to please _not_ use the bugtracker to give non-asked for or misleading information and confuse people with it. You like to use the bugtracker as forum to subtly or not so subtly express disdain for this project. Stop it.
You've helped people with relevant and requested information: please stick to that.

rijotech, 2020-09-11:
Thank you for your concern @daveythacher but I have fixed my screen using the method sugested by mr. @hzeller.

daveythacher, 2020-09-11:
@hzeller Look into inrush current before you cause damage to personal property! You have gaps in your documentation. The Pi was founded as platform to TEACH. You have spread blanket statements.

I would ask that you actually make statements with explanations rather than blanketed generalizations. I also would ask you not to assume what I have or have not done to other people. I would ask that you explain your project in manner that is accurate and conductive to future learning.

I find your claims confusing and I hold a BS in Computer Engineering.