hzeller/rpi-rgb-led-matrix #1146: 128x64 with adafruit bonnet issues

xxxDrewedxxx, 2020-09-11:
I am attempting to get a 128x64 p2.0 running off of an adafruit bonnet. 

Using:
sudo examples-api-use/demo -D0 --led-rows=64 --led-cols=128 --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=4 --led-panel-type=FM6126A --led-multiplexing=0 --led-scan-mode=0 --led-row-addr-type=0 led-chain=1 led-parallel=1

I get this:

If I change the --led-row-addr-type to anything but "0" I get a black screen.
I have pad "E" on the bottom of the bonnet jumpered to "8"

[image]

Basically this, but missing every other row.
[image]

I really am over my head here, so please use small words!  lol
Thanks!

daveythacher, 2020-09-12:
The easiest way to do this is either give me the part numbers of the ICs on the back. Or take a really high quality picture of them to where I can easily make them out.

Ideally I would like pictures off all of it but I am suspecting a row address issue. It is skipping rows so lets start there. The last bullet below:

- Likely there will be 10-100 of one type of IC in lines all over. (ICNxxxx, MBIxxxx, JXIxxxx, FMxxxx, etc.)
- There will be around two ICs near the middle/near the input connector. (xx245)
- There could be one or more additional IC types towards the middle. The number of these will vary. These at the moment are the most interesting for your issue. Usually the bigger caps (big can things) are placed here.

davemaster, 2020-09-12:
Greetings,

-What hardware connections do you have? adapter? which one?
-Also, try your module as two 64x64 modules: led-chain=2, test without --led-panel-type=FM6126A, 
-This option: --led-slowdown-gpio=4 is for Raspberry Pi 3, or 4; start with --led-slowdown-gpio=1,2,3

Best regards

xxxDrewedxxx, 2020-09-12:
It looks like most of the chips are FM 6126B (RED)
There are a column of RT5957DSP down the middle (BLUE)
There are 3 74HC245C in a row running through the middle (YELLOW)
[image]

[image]
[image]
[image]

xxxDrewedxxx, 2020-09-12:
@davemaster 
-What hardware connections do you have? adapter? which one?
RaspberryPi 4,  ADAFRUIT RGB MATRIX BONNET FOR RASPBERRY PI, 
-Also, try your module as two 64x64 modules: led-chain=2, test without --led-panel-type=FM6126A,
led-chain=2 gave me 4 "squares" instead of 2

-This option: --led-slowdown-gpio=4 is for Raspberry Pi 3, or 4; start with --led-slowdown-gpio=1,2,3

daveythacher, 2020-09-12:
You have shift register based row drivers. (Blue) Excellent photos!
Clear the E jumper.

There are shift register address types:
--led-row-addr-type=1 (This one is what I think you want.)
--led-row-addr-type=3
--led-row-addr-type=4 (Not yours)

I think the one you want is --led-row-addr-type=1 but I do not think it is supported.

I would like to ask you to try something. Run it with --led-row-addr-type=1 and take a wire or paper clip and touch C(27) to GND. This pin will hopefully be in tri-state or input, so it should not cause any damage. Reboot Pi before doing this, just as precaution.

Any damage possible to panel would have likely happened under --led-row-addr-type=0. We are only risking GPIO pin 27, which I think is low risk.

Below is for later on if it works: (Ignore this for now)
Datasheet: (RT5957)
http://www.raffar.com.tw/upload_file/2019071017305286848.pdf

Code for --led-row-addr-type=1:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/55fa32fc2e02afb254ac834aea93589d5b891a11/lib/framebuffer.cc#L171-L194

Why I think it does not work:
A is LCK
B is DIN
C is BK (Possibly active low, with pull up.)

C B A
0 0 0
0 0 1
0 1 0
0 1 1
1 0 0
1 0 1
1 1 0
1 1 1

xxxDrewedxxx, 2020-09-12:
So GPIO 27 to ground on the bonnet? With the system running, or power down, jump between 27 and hound, and then boot with it that way?
And remove the solder bridge that I installed on E to 8?

daveythacher, 2020-09-12:
Remove solder bridge on E, disconnect power, count to 10, reconnect power, let it boot, start demo with partially working command changing only address type, then lightly touch jumper between 27 and GND on bonnet, and see if panel lights correctly, report back.

xxxDrewedxxx, 2020-09-12:
--led-row-addr-type=1  -> Black screen.   No change with 27 and ground

daveythacher, 2020-09-12:
That was a long shot. You have voltmeter with continuity?

xxxDrewedxxx, 2020-09-12:
yep

daveythacher, 2020-09-12:
Okay this will be much faster then. Disconnect the panel completely. Use continuity to trace A, B, and C from input connector to pin on U1 and U2. 

If you find the pin on the left side (the side with dot closest too it) then the output pin of interest will be on the right side across down one from dot. If on the right side it will be across up one to dot.

Then see if where the output pin connects on UU1, UU2, UU3, UU4, UU5, UU6, UU7, UU8. Some may not connect at all to all UUx. Two UUx is expected to get one or two connections. I would expect all matches to be on left side of notch near notch.

I am still trying to come up with a reaons --led-row-addr-type=0 works. However this is how I would have done it from the start.

xxxDrewedxxx, 2020-09-12:
Ok, here is what I got.
Yellow are  when the ABC connect.  A and B on U1 and C on U2
Red is the output(?) and where it connects to ALL UUX chips (1-8)
[image]

daveythacher, 2020-09-12:
The bottom right two pins for U1. None of those make it to UU1-8?

Did you try all of these? --led-multiplexing=<0..17>

xxxDrewedxxx, 2020-09-12:
Sorry. Updated.  
Blue and green go to all UUX chips.
Orange only seems to go to UU1
I hadn't gotten though all the multiplexing.  Most of them just seem to make multiple of the same problem.
[image]

xxxDrewedxxx, 2020-09-12:
I've gone though all the multiplexing.  None are any better.  7, 9, 15, and 16 are black screens for me.

daveythacher, 2020-09-12:
Those are awesome pictures. We now know:
A is LCK
B is BK
C is DIN

These means we should have been using --led-row-addr-type=3. I still do not know what value to put on B. We will come back to that.

UUx pin 9 is the pin above the red arrow on UU4.
UUx pin 2 is the pin with the orange arrow on UU4.

I would expect the following:
UU1 pin 9 to go to UU2 pin 2
UU1 pin 9 to go to UU3 pin 2

Does UU3 pin 9 to go to UU4 pin 2 or does U2 pin 12 go to UU4 pin 2?

Ignore all this:
This was the other combination:
C cannot be CLK
A can be CLK
if B is CLK A cannot be data

if A is clock
B DIN -> 01 repeating
C DIN -> 0011 repeating

if B is clock
C DIN -> 01 repeating

However I am still not sure that makes sense to some degree. Except there is only two lines of on the panel? Plus row skipping?

daveythacher, 2020-09-12:
Sorry I forgot to mention U2 pin 12 is the pin with the orange arrow on U2. Do you know IC pin notation, by any chance?

xxxDrewedxxx, 2020-09-12:
Using this numbering
[image]

UU1 pin 10 goes to UU2 pin 2
UU2 pin 10 goes to UU3 pin 2
UU3 pin 10 goes to UU4 pin 2
U2 pin 12 goes to UU1 pin 2 only

daveythacher, 2020-09-12:
Does UU4 pin 10 go to UU5 pin 2? Does the chain move all the way through to UU8?
Sorry about the mix of 9 and 10.

xxxDrewedxxx, 2020-09-12:
Yes, UU4 pin 10 goes to UU5 pin 2, etc all the way through UU8

daveythacher, 2020-09-12:
That is interesting a panel with 1:32 multiplex that is potentially really a 1:64 row scan. Was not expecting that.

You need to define this macro and recompile:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/55fa32fc2e02afb254ac834aea93589d5b891a11/lib/framebuffer.cc#L40

Along with use --led-row-addr-type=3.

I am not sure this explains everything. We still need to figure out if BK is blanking the screen. We also need to figure out how the screen is divided. 

Lets repeat the long shot test with using the correct address mode. It will not work correctly, but hopefully it will light up. 

This is to hopefully address BK. If we get it resolved we can work on the macro, and clean up anything else with a pixel mapper if required due to strange screen division.

daveythacher, 2020-09-12:
Future: (Ignore for now)
Uncomment this:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/55fa32fc2e02afb254ac834aea93589d5b891a11/lib/Makefile#L160

From Top level directory for project run these:
make clean
make all

That should apply the change to ONLY_SINGLE_SUB_PANEL, rebuild library and demo code with new library.

xxxDrewedxxx, 2020-09-12:
Ok, You lost me there.  How do I define the macro and recompile?
Remove the # from line 40? then exit and save?
I'm trying the row add type 3 and GIPO 27 long shot

xxxDrewedxxx, 2020-09-12:
Address type 3 gives black screen.  GIPO makes no difference.

daveythacher, 2020-09-12:
Yes you remove the leading # line, save and exit. (This will define the macro for you.) Move back a directory and run the two make commands. (These commands will activate scripts which recompile it for you, I think.) Then try to run with address type 3. 

Come to think of it that will still not work. We may need a more serious code change. We can worry with that later. I would like to explain why the panel is not coming on with address type 3.

xxxDrewedxxx, 2020-09-12:
I'd like to explain that too, but I haven't a clue!  It only lights up on address type = 0

xxxDrewedxxx, 2020-09-12:
Changed line 40.  no difference.  Didn't change line 160.  Line 160 is uncommented but says "DEFINES+=$ (USER_DEFINES)

daveythacher, 2020-09-12:
Is your power supply shutdown by any chance in address type 3?

xxxDrewedxxx, 2020-09-12:
Ummm.   I don't know.  how would I check that?

daveythacher, 2020-09-12:
I am not sure this is worth our time. Lets just unplug everything. Power supply and Pi. Boot it up running with address type three. Then use a volt meter to measure voltage across main power connector. Power connector has two labels under it VCC and GND. Put the black to the GND outside pin and red to VCC outside pin,

xxxDrewedxxx, 2020-09-12:
0.024 volts as measured on the connector and on the panel
This is the same voltage for address type zero

daveythacher, 2020-09-12:
Just for sanity lets disconnect the power supply from the panel. What the voltage is? What is the power supply power rating?

daveythacher, 2020-09-12:
Wait same for 0? That is not right.

xxxDrewedxxx, 2020-09-12:
Whoops.  I'm  a moron.  DC voltage setting works better.  4.89 vdc.

xxxDrewedxxx, 2020-09-12:
My power supply is 5VDC, 10 Amp

xxxDrewedxxx, 2020-09-12:
Sorry that was my bad.  Had the meter set wrong.

--led-row-addr-type=3   4.98 VDC at the pins on the panel
--led-row-addr-type=0  4.99 VDC at the pins on the panel

xxxDrewedxxx, 2020-09-12:
[image]

daveythacher, 2020-09-12:
That should work, however it may struggle at full power. Voltage is within spec. So not a power problem.

This test is a little risky. I will understand if you want to wait or do something else. 

Those RT5957 are power switches which turn on the LED. The output pins will have some voltage one them if enabled. On each UUx chip there is 8 outputs. Pins 16, 15, 14, 13, 5, 6, 7, and 8.

Now you will not likely see the full 4.96 volts, but you should see something around or above 4 volts if the pin is static. However the pin is not static so you may see some average which is low or bouncing on the meter. (1 to 4.95 volts) If we see any of this then we know the pin driver is enabled.

The goal here is to figure out if a channel is being enabled by the address type 3 logic. If so which ones. We could repeat for address type 0 logic.

To measure it you would use the same location for the black probe. However for the red probe you would would carefully touch each of the listed pins for each UUx IC. Sometimes you need to stab it lightly to get a reading, but make sure you only touch a single pin.

daveythacher, 2020-09-13:
It is possible BK is latch and oe in one. When bk goes low it connects
register to driver. Therefore it does not care about state. It may lock
register input during this condition.

Which could explain why it never worked.

It is another long shot. I while try to mod the code tomorrow, but I am not
able to test.

xxxDrewedxxx, 2020-09-13:
On type 3 logic I get about 0.78~ VDC on pins 16, 15, 14, 13, 5, 6, 7, and 8. on every UUx
On Type 0 zogic I get : 

  | UU1 | UU2 | UU3 | UU4 | UU5 | UU6 | UU7 | UU8
-- | -- | -- | -- | -- | -- | -- | -- | --
16 | 1.15 | 2.7 | 2.5 | 2.6 | 2.8 | 2.7 | 2.5 | 2.7
15 | 1 | 1.1 | 1 | 1 | 1 | 1 | 1 | 1
14 | 2.6 | 2.6 | 2.3 | 2.4 | 2.4 | 2.6 | 2.6 | 2.4
13 | 1.2 | 1.2 | 1.2 | 1.2 | 1.1 | 1 | 1 | 1
  |   |   |   |   |   |   |   |  
5 | 2.5 | 2.5 | 2.6 | 2.5 | 2.6 | 2.5 | 2.5 | 2.7
6 | 1.1 | 1 | 1 | 1.1 | 1.1 | 1 | 1 | 1.1
7 | 2.2 | 2.6 | 2.5 | 2.5 | 2.6 | 2.5 | 2.5 | 2.6
8 | 1 | 1.2 | 1.1 | 1.1 | 1.2 | 1 | 1.1 | 1

daveythacher, 2020-09-13:
This proves the 01 pattern and also possibly a 10 pattern. Which is possibly in line with what I was thinking. I think I would like to proceed with a code change to one of those shifters. I will look into that next.

You can ignore this: (It may not even be useful or correct)
I made a silly little program to try to see if I could model the RT5957. It could predict the following repeating patterns:
Non Concurrent / Non Locking / Active High
e38 and 1c7
Non Concurrent / Non Locking / Active Low
38e and c71
Concurrent / Non Locking / Active Low
c3 and f0
Concurrent / Non Locking / Active High
c3 and f0
Non Concurrent / Locking / Active High
5 and a
Concurrent / Locking / Active High
c and 9 and 6
Concurrent / Locking / Active Low
c and 9 and 6
Non Concurrent / Locking / Active Low
5 and a

Code:
```C
#include <stdio.h>
#include <stdint.h>

int main(int argc, char **argv) {
        uint8_t isLocking = 1;
        uint8_t isBKActiveLow = 1;
        uint8_t isConcurrent = 0;
        uint8_t row, iterations = 10;
        uint32_t regs = 0;
        while (iterations--) {
                for (row = 0; row < 32; ++row) {
                        if ((((row / 2) & 1) == !isBKActiveLow) && !isConcurrent)
                                printf("%x\n", regs);
                        else if (row & 1)
                                regs = (regs * 2) + ((row / 4) & 1);
                        if ((row & 1) && !isLocking)
                                regs = (regs * 2) + ((row / 4) & 1);
                        if ((((row / 2) & 1) == !isBKActiveLow) && isConcurrent)
                                printf("%x\n", regs);
                }
        }
        return 0;
}
```

daveythacher, 2020-09-13:
This is just general documentation for what I am thinking. We can ignore this for now, because it could be wrong. However it will talk about the motivation and type of change being applied.

I think you have shift registers in series. Which have serial clock (LCK) and data (DIN/DOUT). BK clock is output enable and data latch. However the question I keep asking myself was how many register layers does it have and does it keep track of state? This is important because it needs to know when to load data, but it can fall into misaligned states. So assuming the designers are at least semi-sane they would block this.

The answer is the RT5957 can use the BK to lock the serial bus. When set there is no way to update the outputs. So if they default to all off, and we never drive it correctly. Then the outputs will stay off. It will be driven semi-randomly by x to 2^x decoder logic (address type 0).

This means the RT5957 never gives a darn about state. Whatever is in the shift registers at the time of assert BK is correct and can never be altered. Currently to my knowledge there is no proper logic for the RT5957, but we should be able to test this fairly easily.

I think what we will end up doing is driving it like a serial latch or SPI chip/cable/slave select line.

daveythacher, 2020-09-13:
Ignore this:
Actually it is semi supported here: (However we have different setup)
https://github.com/hzeller/rpi-rgb-led-matrix/blob/55fa32fc2e02afb254ac834aea93589d5b891a11/lib/framebuffer.cc#L121-L160

daveythacher, 2020-09-13:
I would like you to find the following lines in lib/framebuffer.cc:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/55fa32fc2e02afb254ac834aea93589d5b891a11/lib/framebuffer.cc#L207-L239
and replace them with these:
```C
class ABCShiftRegisterRowAddressSetter : public RowAddressSetter {
public:
  ABCShiftRegisterRowAddressSetter(int double_rows, const HardwareMapping &h)
    : double_rows_(double_rows),
      row_mask_(h.a | h.b | h.c),
      clock_(h.a),
      bk_(h.b),
      data_(h.c),
      last_row_(-1) {
  }
  virtual gpio_bits_t need_bits() const { return row_mask_; }

  virtual void SetRowAddress(GPIO *io, int row) {
    io->SetBits(bk_);  // Enable serial input for the shifter
    for (int activate = 0; activate < double_rows_; ++activate) {
      io->ClearBits(clock_);
      if (activate == double_rows_ - 1 - row) {
        io->SetBits(data_);
      } else {
        io->ClearBits(data_);
      }
      io->SetBits(clock_);
    }
    io->SetBits(clock_);
    io->ClearBits(clock_);
    io->ClearBits(bk_);  // Disable serial input to keep unwanted bits out of the shifters
    last_row_ = row;
  }

private:
  const int double_rows_;
  const gpio_bits_t row_mask_;
  const gpio_bits_t clock_;
  const gpio_bits_t data_;
  const gpio_bits_t bk_;
  int last_row_;
};
```

Then recompile using the steps before:
Move back to top level directory/folder
Run: make clean
Run: make all

daveythacher, 2020-09-13:
Possibly RT5957 is similar to SM5266P in #981 
https://datasheetspdf.com/pdf-file/1328896/Sunmoon/SM5266P/1

May be similar to LS9739  in #958

xxxDrewedxxx, 2020-09-13:
I ended up with:
make[1]: *** [Makefile:185: framebuffer.o] Error 1
make[1]: Leaving directory '/home/pi/rpi-rgb-led-matrix/lib'
make: *** [Makefile:15: lib/librgbmatrix.a] Error 2

xxxDrewedxxx, 2020-09-13:
~/rpi-rgb-led-matrix $ sudo examples-api-use/demo -D0 --led-rows=64 --led-cols=128 --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=4 --led-panel-type=FM6126A
sudo: examples-api-use/demo: command not found

Whoops!  
Well, crap.  let me go git the files again.

xxxDrewedxxx, 2020-09-13:
Bonus!  No errors when compiling this time.

daveythacher, 2020-09-13:
Does a row light up with --led-row-addr-type=3? After you apply those changes.

xxxDrewedxxx, 2020-09-13:
[image]

[image]

xxxDrewedxxx, 2020-09-13:
Success!!!!

daveythacher, 2020-09-13:
Did you need a code change? What made it work?

xxxDrewedxxx, 2020-09-13:
1. Deleted all Dir in the pi and started from scratch.
2. After no change from before I changed the code in rpi-rgb-led-matrix/lib/framebuffer.cc as you stated
3. Now it works with: sudo examples-api-use/demo -D0 --led-rows=64 --led-cols=128 --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=4 --led-panel-type=FM6126A --led-row-addr-type=3

daveythacher, 2020-09-13:
What caused the red screen?

daveythacher, 2020-09-13:
I think there is still a change needed. Since all the rows are in a single shift register chain I do not thing it is correct. It will not matter a lot based on how the logic works here:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/55fa32fc2e02afb254ac834aea93589d5b891a11/lib/framebuffer.cc#L863-L873

It should be an easy fix. I just wanted to see if this would work first. I looks like we will not need a pixel mapper. Based on the first picture for demo 0.

daveythacher, 2020-09-13:
New logic is:
```C
class ABCShiftRegisterRowAddressSetter : public RowAddressSetter {
public:
  ABCShiftRegisterRowAddressSetter(int double_rows, const HardwareMapping &h)
    : double_rows_(double_rows),
      row_mask_(h.a | h.b | h.c),
      clock_(h.a),
      bk_(h.b),
      data_(h.c),
      last_row_(-1) {
  }
  virtual gpio_bits_t need_bits() const { return row_mask_; }

  virtual void SetRowAddress(GPIO *io, int row) {
    io->SetBits(bk_);  // Enable serial input for the shifter
    for (int activate = 0; activate < (double_rows_ * 2); ++activate) {
      io->ClearBits(clock_);
      if ((activate % double_rows_) == double_rows_ - 1 - row) {
        io->SetBits(data_);
      } else {
        io->ClearBits(data_);
      }
      io->SetBits(clock_);
    }
    io->SetBits(clock_);
    io->ClearBits(clock_);
    io->ClearBits(bk_);  // Disable serial input to keep unwanted bits out of the shifters
    last_row_ = row;
  }

private:
  const int double_rows_;
  const gpio_bits_t row_mask_;
  const gpio_bits_t clock_;
  const gpio_bits_t data_;
  const gpio_bits_t bk_;
  int last_row_;
};
```

xxxDrewedxxx, 2020-09-13:
> What caused the red screen?

Ran a different demo to check all lines active.

daveythacher, 2020-09-13:
Well if you want to try out that latest change that would be great. If not then you are all set. Sorry it took me a while to figure out the RT5957.

daveythacher, 2020-09-13:
@hzeller Here is another RowSetter configuration. Above is a modified version of the ABCShiftRegisterRowAddressSetter.

It includes a few modifications for this panel which is based on RT5957, which is shift register related to SM5266. Unlike the ABCShiftRegister it does use the B line for BK. Unlike the SM5266 it does not have enable signals connected to D and E.

Unlike almost all the address setters it has 1:64 row scan with a 1:32 multiplex. Normally these are equal. Logic was added to correct this. However it may be recommended to use (double_rows_ * SUB_PANELS_) rather than (double_rows_ * 2).

I do not know if you plan to make this its own RowSetter or not. Note we did not end up switching SUB_PANELS_ to 1 via macro in make script.

xxxDrewedxxx, 2020-09-13:
Tried the new code,  got an error when recomplining.

[9 more replies]