hzeller/rpi-rgb-led-matrix #1149: Wrong colors on k716-128x64-32s modules

rijotech, 2020-09-13:
I'm not really sure why but yellow looks pink, blue looks green and green is  yellow on k716-128x64-32s modules, I was wondering if that's something related to the matrix, or if there something I should or can do to correct this problem?

https://imgur.com/a/zvUp00s

hzeller, 2020-09-13:
If you have connected the panel with the regular pin-out and the colors are off it means that the panel internally has wired the RGB colors to different LEDs.
The library has a  way to remap these colors:
You can change which LED is considered red, green and blue with the `--led-rgb-sequence` command-line option. By default, this string is `RGB`; you can give this option a string with a different sequence so you can play with this to see which permutation works for you. `RBG` maybe ?

It is strange that 'green' looks yellow, as this would indicate that a single color would light two types of LEDs. Maybe there is also some LED inversion going on ?
If the LEDs are inverted in their behaviour, i.e. if when they are on they are acutally off, you need the `-led-inverse` option.

rijotech, 2020-09-14:
I will take a look at the possibles solutions, Thanks!

daveythacher, 2020-09-14:
Run demo 5. (Will test primary color function!) Report if RED works under this.

rijotech, 2020-09-14:
WOW yes this worked for me --led-rgb-sequence=RBG thanks again Mr. Zeller.

closed 2020-09-14 by rijotech

daveythacher, 2020-09-14:
https://www.bing.com/images/search?view=detailV2&ccid=Keuh5oGY&id=693BD28A9146496453071D21F9FA6C3CA50CCC83&thid=OIP.Keuh5oGYSaqqdVie-5NbggHaE2&mediaurl=https%3a%2f%2fwww.researchgate.net%2fpublication%2f324229653%2ffigure%2fdownload%2ffig1%2fAS%3a612283747143680%401522991229216%2f3-bit-additive-color-mixing.png&exph=556&expw=850&q=3+bit+color&simid=608009456291152722&ck=1056712D13DB207A75D6D692D3C07BD0&selectedIndex=1&FORM=IRPRST&ajaxhist=0

https://github.com/hzeller/rpi-rgb-led-matrix/blob/97ec43d07f3f85431db4d4a29836fd6a412b0c68/examples-api-use/demo-main.cc#L154-163

https://electronics.stackexchange.com/a/14659

https://www.bing.com/images/search?view=detailV2&ccid=vUN1GIHk&id=636823762A863635D441200C4ACC6F2440C96939&thid=OIP.vUN1GIHkK2zNKbeQ5GOJWgHaHa&mediaurl=https%3a%2f%2fi.stack.imgur.com%2fjwHV9.png&exph=425&expw=425&q=LED+color+change+too+much+current&simid=608042089411510826&ck=F130635127530B605CFD95AE75AEE0B5&selectedIndex=21&FORM=IRPRST

https://github.com/hzeller/rpi-rgb-led-matrix/issues/521#issue-298033947