hzeller/rpi-rgb-led-matrix #1150: Adafruit 3 chained 32x16 flickering issue

Kroutonn, 2020-09-14:
Hello, I'm very new to a lot of this stuff and working with hardware, but I was able to wire up my three 32x16 panels to my raspberry pi 4. Everything seems to be working fine for the most part, but the 3rd panel in the chain seems to have a major issue with flickering pixels. Hard to really capture on camera, but the left panel in the picture is the one with the issue and should just read "Hello" while the other panels properly display.

[image]

I've tried different samples provided, from scrolling images to just printed text, and even switching the panels around with always the 3rd panel in the chain experiencing the flickering issue. Does anyone have any ideas what might be causing this?

daveythacher, 2020-09-14:
It could be power supply issue. It could be configuration issue.

How are the panels wired to the Pi4?
What is your power supply rating? (5V@4A?)
What command/settings did you use to run this?

hzeller, 2020-09-14:
In particular with a Pi4, which is very fast, panels often can't keep up. Try to increase the GPIO slowdown, e.g. `--led-slowdown-gpio=3` or even higher.

Kroutonn, 2020-09-14:
Slowing down the GPIO worked perfectly, thank you so much! And Thank you for all the hard work you've put into this library

closed 2020-09-14 by Kroutonn