hzeller/rpi-rgb-led-matrix #1153: C# Binding: font-example not showing anything

Filipowicz251, 2020-09-15:
Hello

I've seen the issue reported here: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1024

If I understood correctly it has been fixed, right?

But i still have problem running it on mono. I am running it like this (Raspberry PI 4):

` sudo mono font-example.exe /home/pi/led2/fonts/8x13.bdf text --led-slowdown-gpio=2 --led-rows=16`

i am getting only this:

https://i.imgur.com/0bYIfQK.mp4

Clearly, it is not the requested: "text" :)

Additionally, I would want to try it also on dotnet core - but haven't found a way to overcome the "root execution" issue.  I cannot use sudo dotnet (as it claiming that no such command exists) and when i use only dotnet - it is stating that it cannot run this way (because it has to be runned on high privilages). Maybe some tip how to overcome that as well?:)

Ahhh, one additional note. I am using: `--led-slowdown-gpio=2` because i've runned other examples and without  that parameter it breaks.

Final note - my led screen is 16x32 - this one to be exact: https://www.dfrobot.com/product-1643.html

hzeller, 2020-09-15:
Did you verify that the panel works with the regular examples first in `examples-api-use/demo` ?

Other than that: the options structs might have changed slightly with more options, so you might have to double-check that/if the c# binding is doing the right thing, i.e. if the [struct RGBLedMatrixOptions](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/led-matrix-c.h#L48) matches the mapping in C# (the C# binding is essentially maintained by users of c#, so if you find issues, please send a pull request).

Filipowicz251, 2020-09-15:
@hzeller - I've just use the constructor with one RGBLedMatrixOptions (and not with 3 ints) and it is working correctly right now. Thanks for fast anwser!

closed 2020-09-15 by Filipowicz251