hzeller/rpi-rgb-led-matrix #1154: Lower half of the Led matrix displays BLUE instead of GREEN

Filipowicz251, 2020-09-15:
I've tested display of TEST on my led matrix. my c# code is as following:

`        static void Main(string[] args)
        {
            var matrix = new RGBLedMatrix(new RGBLedMatrixOptions() { ChainLength = 1, Cols = 32, Rows = 16, GpioSlowdown = 2, PwmLsbNanoseconds = 250, Brightness = 30});
            var canvas = matrix.CreateOffscreenCanvas();
            while (true)
            {
                Console.WriteLine("Write text to show: ");
                string text = Console.ReadLine();
                canvas.Clear();
                if (text == "q")
                {
                    break;
                }

                var font = new RGBLedFont("8x13.bdf");

                canvas.DrawText(font, 0, 12, new Color(0, 255, 0), text);
                matrix.SwapOnVsync(canvas);
            }
        }`

It is almost working correctly - but i am getting the lower half of the led matrix with shifted color (from Green to Blue) What could be the cause? I am almost sure i've made all the pin connections correctly because i did double (and triple) check it. But would such an effect be caused by mismatch in GPIO wiring?

hzeller, 2020-09-15:
mmh, if you manually wired it, maybe you have to re-wire that second half then (that is the R2, G2, B2 wires) and try the permutation that fixes the color.
If your wiring is correct, then maybe the panel has a strange wiring ?

Filipowicz251, 2020-09-15:
I can try :) But i did make it according to the wiring manual located here:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

One silly question (sorry, this is my first encounter with Raspberry pi and electronic "on this level" :)  tbh) can i hotwire the pins on the fly - or i can break (more like burn) things doing so, and it is better to power off RP and rewire, and turn on? :)

hzeller, 2020-09-15:
If you wired it correctly, this is why I suspect there might be something wrong with panel. But the nice thing is, that is allows you to fix it easily.

If you are careful, it should be possible to hotwire the output to another input, however be aware that the GPIO pins are directly wired to the central CPU, so any screw-up (e.g. shortening to some power rail) might break your Pi. I'd just power it down and do the wiring when it's off.

Filipowicz251, 2020-09-15:
@hzeller - but i've tested it on 2 led panels, that i've ordered. The results is the same. Is it possible to have the same effect on two leds (of course probably from one batch)

hzeller, 2020-09-15:
if you get the panels from the same manufacturer, yes. Sounds like a systematic failure then, as this is the way they wire the PCB.

Filipowicz251, 2020-09-15:
Ok, i DID change B2 with G2 (19 and 21 pin on RP) and it is working correctly;D Go figure ;)

closed 2020-09-15 by Filipowicz251