hzeller/rpi-rgb-led-matrix #1156: Python binding: AttributeError: 'rgbmatrix.core.RGBMatrixOptions' object has no attribute 'panel_type'

robgyiv, 2020-09-18:
First of all, thanks so much for creating such a great library! It's been a lot of fun to be able to start writing some code and have it show on a panel so easily.

Currently on master, after following the readme and building for Python I get the following error:

```bash
$ sudo ./samples/pulsing-colors.py
Traceback (most recent call last):
  File "./samples/pulsing-colors.py", line 41, in <module>
    if (not pulsing_colors.process()):
  File "/home/pi/code/github.com/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py
", line 56, in process
    options.panel_type = self.args.led_panel_type
AttributeError: 'rgbmatrix.core.RGBMatrixOptions' object has no attribute 'panel_type'
```

I have set `HARDWARE_DESC=adafruit-hat` in `lib/Makefile`.

If I checkout to commit 4c9a4036f44f and rebuild this does not happen. There may be a more recent commit that also works but I haven't tried it.

I'd be happy to contribute to the issue (more so in Python than C++) but may need some direction first.

Thanks again and let me know if any more info would be useful.

Hardware:
- Raspberry Pi 2
- [Adafruit RGB Matrix Bonnet](https://learn.adafruit.com/adafruit-rgb-matrix-bonnet-for-raspberry-pi?view=all)
- [LED panel](https://learn.adafruit.com/32x16-32x32-rgb-led-matrix?view=all#faq)

drewlustro, 2020-10-25:
I am also experiencing this! `master` appears to be broken

cpforbes, 2020-10-29:
It looks like commit 56b87b98f8aea9a95ee93449cf00831bcb3f0efa broke the python bindings.

robgyiv, 2020-12-18:
Thanks @cpforbes - this was the case for me as well as I used the adafruit guide initially. So the fix for me was:

- Removing `rgbmatrix` from `/usr/local/lib/python(2|3)/dist-packages`
- Rebuilding from the Python bindings in this repo

closed 2020-12-18 by robgyiv