hzeller/rpi-rgb-led-matrix #1157: difficulty making a first easy sketch by myself with c++

carmonkeylab, 2020-09-20:
hi guys, as the title suggests, I am having trouble creating the sketch by myself. do i need to compile the sketch, if so, how would i do that. where would i need to create the sketch. do i need it to be inside the include library for the library to properly work. also, if possible, can someone send basic code that just fills the screen a solid color with comments on most, if not all lines so i can properly understand what each command would do. thanks for the help in advance

carmonkeylab, 2020-09-20:
forgot to state that i am using a raspberry pi 4

daveythacher, 2020-09-20:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/97ec43d07f3f85431db4d4a29836fd6a412b0c68/examples-api-use/demo-main.cc#L1063-L1064

https://github.com/hzeller/rpi-rgb-led-matrix/blob/97ec43d07f3f85431db4d4a29836fd6a412b0c68/examples-api-use/demo-main.cc#L1066-L1068

https://github.com/hzeller/rpi-rgb-led-matrix/blob/97ec43d07f3f85431db4d4a29836fd6a412b0c68/examples-api-use/demo-main.cc#L1071-L1075

https://github.com/hzeller/rpi-rgb-led-matrix/blob/97ec43d07f3f85431db4d4a29836fd6a412b0c68/examples-api-use/demo-main.cc#L1102-L1104

https://github.com/hzeller/rpi-rgb-led-matrix/blob/97ec43d07f3f85431db4d4a29836fd6a412b0c68/examples-api-use/demo-main.cc#L89

https://github.com/hzeller/rpi-rgb-led-matrix/blob/97ec43d07f3f85431db4d4a29836fd6a412b0c68/examples-api-use/demo-main.cc#L60

https://github.com/hzeller/rpi-rgb-led-matrix/blob/97ec43d07f3f85431db4d4a29836fd6a412b0c68/examples-api-use/demo-main.cc#L82-L83

https://github.com/hzeller/rpi-rgb-led-matrix/blob/97ec43d07f3f85431db4d4a29836fd6a412b0c68/examples-api-use/demo-main.cc#L1186

carmonkeylab, 2020-09-20:
>do i need to compile the sketch, if so, how would i do that. where would i need to create the sketch. can i put it wherever i want or am i supposed to put it somewhere designated.

carmonkeylab, 2020-09-20:
can you pls answer these 2 questions

closed 2020-09-26 by carmonkeylab