hzeller/rpi-rgb-led-matrix #1161: black two rows

chananegan, 2020-09-26:
Hi,

This is Paul. I have been really impressed by this rgb led matrix projects and  trying myself to apply this matrix for decorating my living room. I am almost new to dealing with electronic hardware. 

I am using rpi 3 and rpi 3 B+. 
I purchased two adafruit rgb matrix bonnet and one HAT. 
I also purchased adafruit 32x32 rgb matrix and 64x32 rgb matrix. 

The reason I had purchased this many materials is that I failed in turning on the matrix during trials and errors, and was not sure what was the problem. 

Testing with the materials I succeeded turning on the matrix (both 32x32 and 64x32) with one of the two adafruit bonnets. However, I did not make it with the other bonnet and HAT. 

On the terminal I used: **sudo ./demo -D0 --led-gpio-mapping=adafruit-hat --led-cols=32 --led-slowdown-gpio=2**

The problem with the bonnet and the HAT is like this:
[image]

[image]

The strange thing is that I made it with one of the bonnets but the other bonnet and HAT did not do the job and the two of them have exactly the same error: **blank two lines**

The error was happening with both 64x32 and 32x32 matrices in the same way. I tried with adding option commands like "types of display, led multiplexing, led row addr type and so on" but with no luck.

If I am told by someone that I accidentally bought bad bonnet and HAT with same defects that happened in manufacturing, soldering or something, I would gave up, throw them into trash bin and forget about them, but I think it is very little possibility to happen like this and still wondering if there is the other way around. Reading and searching for similar issues here I decided to make a new issue. 

Thank you!
Paul

carmonkeylab, 2020-09-26:
i think i remember having the same problem. try this "If you're using a 64x64 RGB matrix and either a Bonnet or a Rev C HAT, use your soldering iron to melt a blob of solder on the bottom solder jumper so the middle pad is 'shorted' to 8. (This is compatible with 64x64 matrices in the Adafruit store. For 64x64 matrices from other sources, you might need to use 16 instead, check the datasheet.)" from https://learn.adafruit.com/adafruit-rgb-matrix-bonnet-for-raspberry-pi/driving-matrices

if i remember correctly, that solved my problem and might solve yours. gl :)

carmonkeylab, 2020-09-26:
you said it is only happening with 64x32 and 32x32 displays

carmonkeylab, 2020-09-26:
if yes then i don't know why. if it was a 64x64 this might help "If you're using a 64x64 RGB matrix and either a Bonnet or a Rev C HAT, use your soldering iron to melt a blob of solder on the bottom solder jumper so the middle pad is 'shorted' to 8. (This is compatible with 64x64 matrices in the Adafruit store. For 64x64 matrices from other sources, you might need to use 16 instead, check the datasheet.)" step 3 from https://learn.adafruit.com/adafruit-rgb-matrix-bonnet-for-raspberry-pi/driving-matrices

if i remember correctly, that solved my problem and might solve yours. gl :)

chananegan, 2020-09-26:
Yes, I only have 64x32 and 32x32 panels. I saw open and closed issues with 64x64 here and knew for 64x64 I heard additional soldering is necessary, but it is not my case. Thank you anyway for taking your time reading my issue and giving advices. Have a good day!

carmonkeylab, 2020-09-27:
np, and good luck

daveythacher, 2020-09-27:
Here are two test options for figuring out if there is damage to the adapters. I would recommend starting with option two then moving to option one. I agree the odds of manufacturing error here are low, but two seems even lower.

Test Option 1:
Connect a LED to the output. Then connect something to the input. Verify that you can blink the LED off and on. Do this for every pin pair. The panels are basically an array of logic buffers. Adafruit documents their schematics which will tell you which input pins map to output pins.

Test Option 2:
However before that you could use voltmeter to test for input and/or output shorts. Look to see if any input pin is shorted to another input pin. Then test the outputs. This is continuity testing so only inputs to inputs and output to outputs.

chananegan, 2020-09-27:
As soon as seeing your reply, I took out my multimeter and check connectivity testing with option 2 and 1 in order, and it seems like the connection has no problem. 

While testing connection, I felt like a slave of Adafruit adapters (bonnet and HAT) because without them I can make the circuit work with certainty even though I have to do some labor with checking pin ins and outs. 

If my bonnet and HAT have actual defects -- God knows -- what did adafruit did to me? Making me time consume and taking money? Really? I hope this is not the case .... haha

Thank Dave for advices.

daveythacher, 2020-09-27:
I am Davey or David, but not Dave, Davy, Davie, etc. Your post was unclear. Did the adapters pass the tests? As far as I can tell you have no real issue other than thinking a few adapters are bad. Technically this inquiry should be directed at Adafruit.

You said the same panels work with one adapter and not two other adapters. This would suggest an issue with the two adapters assuming you used the same library settings. The working one rules out the panels, library, cables, power supply, settings, etc.

I would expect the adapters to work, this is a fairly straight forward test. It is kind of annoying however, based on what you have said it seems appropriate. It seemed to me you were interested in keeping them if functional. Here is a few way to try that. Again looking at the schematics they are basically buffers.

Another way to look at this is that two are correct and one is wrong. To know this I would suggest mentioning the ICs on the back of the panels. However I strongly suspect these are serial shift registers and decoder row select based boards. Every board I have bought from them has been that way. Also these are simpler to work with. All their logic/tutorials is based around this.

chananegan, 2020-09-27:
Yes Davey, I totally agree with your point and I thought first I have to discuss this issue with Adafruit but to me it looked like to me that they are not so eager to hear from a customer after purchase, so here I saw more helpful comments. 
Thanks for analysis!

closed 2020-09-27 by chananegan