hzeller/rpi-rgb-led-matrix #1162: animation with python binding

carmonkeylab, 2020-09-26:
hi guys, i am using the python binding and i am wondering what the easiest way to make animations is. thanks in advance

raspberry-pi-maker, 2020-10-03:
I've posted several blogs about using @hzeller 's library and Raspberry Pi.
I hope my blog(https://iot-for-maker.blogspot.com/2020/04/led-multilingual-led-matrix-output.html) might help you.

carmonkeylab, 2020-10-03:
thanks for sending the link to the blog, but after looking at the blog i saw that you were talking about how to run korean language. i was trying to know what the easiest way to animate a simple character move was but thanks for reading my post anyway

raspberry-pi-maker, 2020-10-04:
If you run the code, the text might flow to left. If you see the source code, pos value decrease by 1  every loop

while True:
    offscreen_canvas.Clear()
    len = graphics.DrawText(offscreen_canvas, font, pos, 25, textColor, my_text)
    pos -= 1
    if (pos + len < 0):
        pos = offscreen_canvas.width

    time.sleep(0.05)
    offscreen_canvas = matrix.SwapOnVSync(offscreen_canvas)

closed 2020-10-05 by carmonkeylab