hzeller/rpi-rgb-led-matrix #1163: Python Samples Errors

mysecretstache, 2020-09-27:
Hello,  I've compiled and the API demo is working fine, but when I try to run the python samples, I'm getting an error message.

> Traceback (most recent call last):
>   File "./runtext.py", line 3, in <module>
>     from samplebase import SampleBase
>   File "/home/pi/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 7, in <module>
>     from rgbmatrix import RGBMatrix, RGBMatrixOptions
> ImportError: cannot import name RGBMatrixOptions

I'm getting this with all the python examples.

hzeller, 2020-09-27:
When you [installed the Python binding](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python) as a pre-requisite to use the Python binding, you also chose which Python version to use. Make sure to run the examples with the same Python version.

mysecretstache, 2020-09-27:
Got it.  I though I had installed for 2.7 & 3.  when I forced to run with python 3 it is working now.  thank you

marcmerlin, 2025-11-30:
I'm told this is resolved now