hzeller/rpi-rgb-led-matrix #1164: When I run video-viewer, the LED panel does not display.

6010110254, 2020-09-28:
i see from >> https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils
i have video name avenger2.webm  1 minute
[image]
[image]
[image]
led 64*32  chain 4 
When I run video-viewer, the LED panel does not display.
[image]

6010110254, 2020-09-30:
Help me pleas @hzeller

hzeller, 2020-09-30:
You didn't say how you connect your pi, but it looks like from the picture in the background it is an Adafruit Hat.

Did you provide the [`--led-gpio-mapping=adafruit-hat`](https://github.com/hzeller/rpi-rgb-led-matrix#changing-parameters-via-command-line-flags) option to switch to that output mapping ?

6010110254, 2020-09-30:
I have --led-gpio-mapping = adafruit-hat     
using
HARDWARE_DESC = adafruit-hat make

Which when I run Image Viewer  ->  sudo ./led-image-viewer I can run it, work normally.
[image]

6010110254, 2020-09-30:
Is there a 3-step process for running a video?
1.
sudo ./video-viewer  myvideo.webm
2.
./video-viewer  myvideo.webm -O / tmp / vid.stream
3.
sudo ./led-image-viewer  /tmp/vid.stream

hzeller, 2020-09-30:
I would always provide the `--led-gpio-mapping=adafruit-hat` option on the command line, that way you are sure that you're using it. Otherwise you have to always give the `HARDWARE_DESC` compile time option, e.g. when you compile the video viewer.

These are two ways to watch a video:

 * can run a video directly (one step) with `sudo ./video-viewer ...`
 * You can pre-compute the video to a stream (`video-viewer -O ...`) and watch with led-image-viewer (a two step process, but the first step you only have to do once).

Given that the decoding of a video uses more resources, you get better quality when doing it the pre-computing way.

Also, if you want the best possible quality, I also strongly suggest to do the [PWM modification](https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker) on the HAT.

closed 2020-10-01 by 6010110254