hzeller/rpi-rgb-led-matrix #1169: Flicker due to USB-serial connection

RobbertGroenendijk, 2020-10-05:
Hi there,

I'm using a Raspberry Pi 4 connected to a Teensy 4.0 by a USB cable connected to tty0 on the Pi. My screen runs all smooth but just get a flicker when the Pi is writing bytes to the Teensy (very short, but noticeable) I'm running Buster Lite. All examples run smooth, and i've tracked my problem down to only something with this serial connection. Is this a known issue, and if so, does anyone have pointers on how to fix it? My code is written in C++.

Cheers!

arahasya, 2020-10-05:
--led-limit-refresh=150-300

daveythacher, 2020-10-07:
There is a lot that can cause this. Power, cache, system bus design internal to processor, etc. I would be guessing without more information.

RobbertGroenendijk, 2020-10-07:
So I ended up actually fixing it, and it was due to something else in my code. With my current settings there is no more flickering due to the USB connection (more people seem to have problems caused by this).
Just for future information these are my current settings;

```
    options.hardware_mapping = "regular";
    options.pixel_mapper_config = "U-mapper";
    options.rows = 32;
    options.cols = 64;
    options.chain_length = 8;
    options.parallel = 3;
    options.show_refresh_rate = false;
    options.pwm_lsb_nanoseconds = 100;
    options.pwm_dither_bits = 1;
    options.scan_mode = 0;
    
    runtime_options.drop_privileges = 0;
    runtime_options.gpio_slowdown = 3;
```

I'm running 24 screens of 62x32 pixels on a Raspberry Pi 4 (4GB ram) connected to a Teensy 4.0 by USB. So if anyone finds this thread and has problems with a similar setup you can send me a message :)

closed 2020-10-07 by RobbertGroenendijk

daveythacher, 2020-10-07:
That is a decent amount of power, about 138.2-184.3 Amps possibly. You are not getting 11 bits of PWM. GPIO slowdown of 3 for Pi4, not sure about that.

RobbertGroenendijk, 2020-10-07:
> That is a decent amount of power, about 138.2-184.3 Amps possibly. You are not getting 11 bits of PWM. GPIO slowdown of 3 for Pi4, not sure about that.

Ye, it's drawing quite a bit of power but for it's fine at the moment for my implementation. I do not have fast-moving images on there and most of the pixels don't even need updating most of the time. I only need 8 bits of PWM. But would you recommend anything else seeing your comment? I've tried GPIO slowdown 2, but this just causes it to flicker so went with this for now.

daveythacher, 2020-10-07:
The current I mentioned is the peak current. The panel has two rows per parallel chain active at a time under most panels. This means you can have at most 64 * 8 * 2 * 3 RGB LEDs on at a time. This can cause issues for power supply if it is not strong enough to push this power. Wiring can also become an issue at current this high.

The worst case update in terms of frequency happens beyond your control in the background. You can work with huge panels if you keep the number of LEDs on in a single row down. A solid white screen will active all LEDs and cause max load. The rate of image/pixel change is not a huge deal since it multiplexes anyways.

You can limit the FPS which will slow it down. However the multiplexing and BCM is out of your control unless you edit the code. The Pi4 is a little faster than the other Pi's which causes it to violate the panel's shift register timing. This library uses the GPIO slow down to inject delays into update process. Usually you see values of 4 or 5 on the Pi4, however it can depend.

You would want to set it for worst case. If the other processes slow down the update process you can possibly get by with lower values. However when/if those processes idle down you could see instability again. Likewise if you exceed the power supply by turning on too many LEDs you can see instability. If you do a lot of stressful computation on the processor you can see instability.

I seriously doubt you are getting more than 5 bits of PWM. This project will technically send 11 bits of PWM. However it will not do it correctly given certain settings. The panels this library supports are based on shift registers with a max clock of 25MHz. This puts limit on the display length you can get using certain bit depths. Little documentation exists to explain this. I tried to bring this up but I doubt it will be addressed.

daveythacher, 2020-10-08:
I think you should not see much of a difference with:
options.pwm_lsb_nanoseconds = 640;
options.pwm_bits = 5;

https://github.com/hzeller/rpi-rgb-led-matrix/issues/1120#issuecomment-705272537

K0rkunc, 2021-06-11:
> Sonunda onu düzelttim ve bunun nedeni kodumdaki başka bir şeydi. Mevcut ayarlarımda, USB bağlantısı nedeniyle artık titreme yok (daha fazla insanın bundan kaynaklanan sorunları var gibi görünüyor).
> Sadece gelecekteki bilgiler için bunlar benim mevcut ayarlarım;
> 
> ```
>     options.hardware_mapping = "regular";
>     options.pixel_mapper_config = "U-mapper";
>     options.rows = 32;
>     options.cols = 64;
>     options.chain_length = 8;
>     options.parallel = 3;
>     options.show_refresh_rate = false;
>     options.pwm_lsb_nanoseconds = 100;
>     options.pwm_dither_bits = 1;
>     options.scan_mode = 0;
>     
>     runtime_options.drop_privileges = 0;
>     runtime_options.gpio_slowdown = 3;
> ```
> 
> USB ile Teensy 4.0'a bağlı bir Raspberry Pi 4 (4GB ram) üzerinde 62x32 piksellik 24 ekran çalıştırıyorum. Yani bu konuyu bulan ve benzer bir kurulumla ilgili sorun yaşayan varsa bana mesaj gönderebilir :)

hi can u share codes me ?