hzeller/rpi-rgb-led-matrix #1175: weird things happening with 64x64 matrix and rpi4

carmonkeylab, 2020-10-11:
hi, i am using an rpi4 and a 64x64 matrix. i used to be running Raspbian lite and everything was good. then i decided to switch to dietpi since it was lightweight. i installed the repo but when i run `sudo ./demo -D0 --led-rows=64 --led-cols=64 --led-slowdown-gpio=5`, this happens 
[image](yes, i broke a small part of the matrix) it does that and sometimes it does something similar to what it does in the pic. also the photo coloring is a little off. i have suspected it was probably a IDC cable problem so i am using going to buy a new one. the reason i suspect it is a cable problem is because sometimes when the cable wiggles a little the display shows nothing. is there any other way to trouble shoot the matrix and make sure it is the cable and not anything else.

daveythacher, 2020-10-11:
On the back side of the panel there are ICs, lets get a list of them in the comments.

carmonkeylab, 2020-10-11:
ok, but the weird thing is that it was working and then stopped working

carmonkeylab, 2020-10-11:
[image]

Red is ICND2038S a24268ja
Blue is mw245b
Green is ruc7258
Purple is 74hc048

i will also send better quality pictures

carmonkeylab, 2020-10-11:
[image]
[image]
[image]

daveythacher, 2020-10-12:
I see no issues with the command you used against the panel configuration. 

To test cable use power LED or use continuity setting of voltmeter. 

Make sure the clamps are fully compressed on the top of each connector of ribbon cable. Make sure the connectors are fully seated in sockets. Make sure the connections to the socket on the pi are stable.

daveythacher, 2020-10-12:
Do you have the original image to jump back to?

You are using Pi4 which is pretty fast compared to the other Pi models. It could be GPIO slowdown of 5 is too low with dietPi. You will need to hack the code to fix that if it ends up being the case.

carmonkeylab, 2020-10-12:
i ended up reinstalling raspbian lite but the same thing happens there

carmonkeylab, 2020-10-12:
i will check the continuity

carmonkeylab, 2020-10-12:
no problem with continuity

carmonkeylab, 2020-10-12:
> Do you have the original image to jump back to?
> 
> You are using Pi4 which is pretty fast compared to the other Pi models. It could be GPIO slowdown of 5 is too low with dietPi. You will need to hack the code to fix that if it ends up being the case.

no i do not have the original raspbian lite image but i put a new fresh one which results in the same thing happening

carmonkeylab, 2020-10-12:
[image]
[image]

i measured the distance between the bonnet and the boards and gave the pics so you can tell me if that is good or if it is supposed to go down more

hdm, 2020-11-11:
Hello! I am running into the same issue. Using the 64x64 with the RGB bonnet on an RPi 4B+ with the default gpio mapping shows the same two green lines as the original poster. Switching to the `adafruit-hat-pwm` configuration gets closer, but its still wrong:

[image]

hdm, 2020-11-11:
Testing the same hw/code/pi image on a 3B+ has a similar green bar in `adafruit-hat-pwm` configuration and is completely broken (but in a different way) with the regular gpio mapping. Maybe something changed with the RGB panels?

hdm, 2020-11-11:
It may be worth noting that the 64x64 has an E address and it looks like this library doesn't have explicit support for that yet.

daveythacher, 2020-11-12:
The library supports E drive. It has to be enabled in the hat with a jumper if I am not mistaken. Also you need to use a pin mapper which supports it. I think the Adafruit pin maps do. The main readme mentions how to set that up.

I think the Pi4 can do weird stuff. The Pi3 is a little more stable as I understand it. I have never worked with either, so take that with a grain of salt.

daveythacher, 2020-11-12:
> Hello! I am running into the same issue. Using the 64x64 with the RGB bonnet on an RPi 4B+ with the default gpio mapping shows the same two green lines as the original poster. Switching to the `adafruit-hat-pwm` configuration gets closer, but its still wrong:
> 
> [image]

That solid green patch would make me look into the green data line responsible for that half of the panel. It looks like it is stuck. Other than that it looks like it is working.

hdm, 2020-11-12:
Thank you for the reply! It looks like this may be a known (if not common) defect in the RGB bonnet boards.
Reading through the Adafruit support forums I found someone with a similar issue:
 - https://forums.adafruit.com/viewtopic.php?f=47&t=170770&p=838423

I posted a similar entry at:
 - https://forums.adafruit.com/viewtopic.php?f=47&t=171713

The "green lines" issue from the original poster matches the behavior I see if the gpio pinout is set to "regular" instead of "adafruit-hat" (or the pwm variant). My guess is that if they switch to `--led-gpio-mapping=adafruit-hat` they will see a similar issue to my photo above.

daveythacher, 2020-11-12:
Write python script to toggle green data pins. Connect LED to green data pins. Run script with nothing else running. 

First try connecting LED directly to Pi. (Remember current limiting resistor.) Then connect LED to HUB75 green data pins.

You can use multimeter if you do not want to use LED and resistor.

This test is designed to see if the bonnet is bad. Looking for location of error in bonnet. Pi side of 74HC245 or panel side. Also if Pi side we can test for issue with configuration, see if something else is trying to drive that pin.

hdm, 2020-11-13:
Thanks for the reply! I went down the list with a multimeter after booting the Pi 4, but with no demo (or other LED program) running:

A/B/C/D/E: All zero as expected
R1: 3.14
R2: 0
G1: 0
G2: 3.99
B1: 0
B2: 0

It sounds like G2 should not be almost 4v while idle.

daveythacher, 2020-11-13:
Did you try to toggle the Gx pins to see if it would wiggle?

daveythacher, 2020-11-13:
If stuck at 4V that would explain the issue. That will likely be read as high which would explain the green bar.

The point is prove if there is a stuck fault in the PCB or 74HC245. This could be due to a lack of solder, short or defect. It could be something else is trying to drive that pin somehow in the Pi. 

The Pi should not be generating 4V from GPIO. These are buffers. So technically you would want to see what the voltage of the input pin is. So where G2 hits the Pi header. Is that 2.64V, 3.3V or 0V? We can make no statement if it is 2.64V. Weak statement can be made if 3.3V. However 0V there should be no 4V.

Point there is prove a relationship that models itself as a buffer exists. If it does not then the problem is between Pi header and HUB75 connector. (Within bonnet.) Otherwise we cannot say an error does or does not exist within the bonnet.

hdm, 2020-11-14:
Thanks again for the reply. I wasn't sure which pins on the Pi mapped to G2 and found something interesting while trying to determine this. It appears that the G2 pin is shorted to quite a few other pins on the board, including the 5V line. A continuity test on G2 hits on about 6 pins. I tested w/o the Pi connected and still see a green line and up to 5V high on the G2 pin (before it goes to the chip). 

Broken PCB?

daveythacher, 2020-11-14:
Below the main Pi header is a row of marked pins. There is a pin marked G1 and G2. It looks like it also marks the Pi GPIO number. So for G1 this is GPIO 13, and for G2 this is GPIO 16. That is the Pi side.

In the center there is two ICs which are buffers. Along with their corresponding traces. If there is a defect here we should be able to spot it with a few quick tests. Personally I would just pull out the schematics and start checking traces. Lets just avoid that for now.

On the HUB75 side. Make the notch on the left. The G1 will be in the upper right pin. G2 will be third from the top on the right.

Lets start with measuring the voltages of those four pins with nothing running. Then lets try toggling those pins very very slowly using a script with nothing else running.

daveythacher, 2020-11-14:
You did the continuity testing disconnected from Pi without power right. That can cause weird stuff to happen. Sometimes for the power rails the meter will beep a little indicating that.

So this could be dead due to a number of reasons. The PCB has defect. The IC has defect. The Pi has defeat. The soldering has defect. (Couple look like cold join but that probably not the issue. Lead free looks like that sometimes.) Maybe even other things.

There is a number of ways to test. These test result follow divide and conquer. The results are designed to create one issue to be more probably than the other. Then you follow that so long as it is true/valid. What I am trying to advice is basic troubleshooting for possible board failure.

You could just try and buy another board. If it works maybe you just toss the other. How effective this is depends on the odds of the boards failure. Which even if semi common is not likely but still possible with too boards. This will not work if the Pi is the culprit.

hdm, 2020-12-30:
Just to close the loop on this, Adafruit sent me a new bonnet and this resolved the issue.