hzeller/rpi-rgb-led-matrix #1178: Reading external signals / trying input-example

Turmburger, 2020-10-14:
Sorry for bothering you with beginner's questions: 
- where do I find the bridge between the GPIO Pin numbers of the code and the physical numbers of the raspberry 3B+?
- when running "input-example": why does no LEDs on the 64 x 64 matrix changes, if I connect open pins (not used by the matrix) with 3,3 V or ground

Thanks you very much for your support
Joachim

daveythacher, 2020-10-15:
The Pins mapping to GPIO pin numbers is found here:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c

Multiple configurations are defined there. The configurations vary in terms of connector number, 1 to 3. Some configuration use address line D and E. Each connector is defined to have two data sets (R, G, B). However some panels use four, so you would use two connectors for those. The OE, CLK and LAT signal are shared between the connectors. By default regular is used. Instructions for editing or selecting a different exist in the main README.md.

These panels work off serial shift registers. Therefore the LEDs may not come on via random toggling of pins. You need to clock data out using data pins, CLK and LAT. Also by default the OE will likely try to disable the LEDs.

Turmburger, 2020-10-16:
Thanks, What are the LEDs in "minimal-example" exactely show? 13 LEDs are on (from 1 to 64): 2, 27, 28, 30, 31, 32, 34, 59, 60, 62, 63, 64). How are this LEDs referring to the GPIOs of the RP?

daveythacher, 2020-10-16:
Can you list IC part numbers on back of LED panel.

Turmburger, 2020-10-16:
- 
[image]
[image]

Turmburger, 2020-10-16:
This are the two main chips

daveythacher, 2020-10-17:
Are you no longer having issues? You closed the issue. Both of those ICs should work. These are not special cases. Do you have adafruit hat?

Turmburger, 2020-10-17:
No, it was not intended to close the issue. Thanks for the hint.
In the meantime I think I understood a part of the problem:  LEDs 2, 27, 28, 30, 31, 32, (left half) = 34, 59, 60, 62, 63, 64 (right half of the display). But it is still not clear for me, what the  LEDs 2, 27, 28, 30, 31, 32 show exactely.

daveythacher, 2020-10-18:
Can you list your command? Including all options you specify.

Turmburger, 2020-10-18:
I reinstalled the repository and got the following results:

Input-example without any options
[image]
[image]

Input-example with options:
[image]
[image]

daveythacher, 2020-10-18:
Now I understand what you are referring to. Sorry about that, I was slow.

How do you have the panel connected? What panel pins connect to which Pi GPIO pins?

I would recommend using a different example program. That program reads GPIO pins to set pixels on the screen.

I would switch over to:
sudo ./demo -D0 --led-cols=64

I am guessing a single panel of 64x32?

Based on the IC you do not likely need to change the multiplexing, address type or panel type. So far I am guessing on the cols, rows, chain length, parallel count, gpio slow down, pwm-bits, pwm-lsb-nanoseconds, etc. You may however need to use a different gpio mapping.

daveythacher, 2020-10-19:
Oh okay. I did not realize you had this already working. The input example runs a loop polling the GPIO bank 1 register. Any pins not used by this project are likely sampled. 

It attempts to read all of them at first. However it may attempt to detect which are configured as output vs input. In any case the pins if determined as inputs are sampled.

GPIO bank 1 pins map to the location of the bit in the register. These bit locations are mapped to the pin header. So the GPIO pin number is the bank bit number. This project only used one of the two banks on the Pi to my knowledge. I think only the compute modules support using the other GPIO bank.

Here is schematic for Pi Zero 1.3 showing the GPIO bank 1:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_Zero_1p3_reduced.pdf
You can also see the logic level used by that bank is 3.3 volts. There was probably a number of values they could have used here. (1.8 is used for analog functions, if my memory is correct.)

Here is a section from input-example:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/b8c7cc42e7f5a852ba93c3cc70e4b31e576632c5/examples-api-use/input-example.cc#L48-L57
On Line 55 you can see it only lights one line. It also accounts for the shift register by reversing the number range.
On Line 50 it samples new values.
On Line 54 it decides if the pixel is on or off. If on it shows white or black if off.
On Line 53 it scans through the 32 bit register value which is the max size of a single GPIO bank.

daveythacher, 2020-10-19:
Why do you need this just out of curiosity?

Turmburger, 2020-10-20:
I want to influence the speed of a video by an external trigger

Turmburger, 2020-10-20:
After trying all GPIO inputs which are not used by the matrix, I found a couple of inputs which influence the leds. So the first part of the mission seems to be completed. The second question is how I can reduce / influence the speed of the video in "video-viewer".

Turmburger, 2020-10-22:
I think, I got it - thanks for support

closed 2020-10-22 by Turmburger