hzeller/rpi-rgb-led-matrix #1179: dsPIC new implementation of RGB matrix ????

gege13007, 2020-10-15:
Hi guys ! I 'm thinking about implementing this great code from hzeller on a small dsPIC microcontroler. A dsPIC30 or dsPIC33 can run to 30-40MIPS or more...
So no more need of a RaspBerry and no need to wait 1-5 seconds on startup. This could be fun for some small applications.
Somebody has tried ?

daveythacher, 2020-10-15:
This is off topic.

However you will end up rewriting a ton of this logic. dsPIC has a lot to offer that this code base and platform conceptually does not have. That being said there could be a few issues. It would be hard to know without more details about setup and controller.

BCM would need to be used. PWM in the normal sense is just to hard for software. Normally you would use a FPGA to work that out. You could precompute and store in RAM then play it back. Interfacing may be a slight challenge, however DMA could help. You will want to use interrupts more than likely.

There is a lot of little games you can play. Once the image is converted into memory it becomes a timing control and bandwidth problem. Depending on frame draw frequency the CPU clock will not be an issue. However you could do this remotely and just copy it over from interface straight to memory.

However that memory will be raw. You could use a few cycles to decompress it. This would save interface bandwidth. The Pi has practically unlimited memory. These controller do not. However the Pi has issues with timing stability which these controllers probably will not.

Lowering you pwm_bits will help with memory quite a bit. dsPIC is limited in external memory bandwidth to my knowledge. This is not true for other controllers. Most move to FPGA which allows huge numbers of connectors and multiple parallel memory chips. Solving the memory size and bandwidth issue. This allows for high performance, low cost, and high quality.

The reason for this is simple. It is repeatable IO task. The generic nature is actually really wrong. However the cost of something like that is much more. This project mostly supports serial shift register LED drivers, there are others which are likely more friendly. The memory mapped LED drivers will store the image internally. You could make huge panels with limited resources on those.

I could go on about this for a long time. dsPIC to my knowledge does not support C++. However I am not sure I would recommend it here. Look into the CH dsPIC, those are awesome.

I was considering making a SPI to HUB75 bridge using a microcontroller. Couple different models considered. However it would likely need external logic to draw frame. That could be a Pi or another microcontroller. 

I would try to get the SPI bandwidth down for multiple channels. Granted the Pi has two SPI ports. I think I started to steer over toward the PIC32 since it has faster SPI ports.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin