hzeller/rpi-rgb-led-matrix #1180: --led-show-refresh shows values 33% slower with HEAD compared to

marcmerlin, 2020-10-22:
Older code:
root@rPi4:~/rpi-rgb-led-matrix# ~/rpi-rgb-led-matrix.sav-faster/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128  --led-row-addr-type=0  --led-show-refresh --led-slowdown-gpio=2   --led-pwm-bits=9 --led-panel-type=FM6126A  --led-parallel=3 --led-pwm-dither-bits=2 --led-pwm-lsb-nanoseconds=100 -D0                        279.1Hz up to 331.0Hz (mostly 330Hz)

Current code:
root@rPi4:~/rpi-rgb-led-matrix# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128  --led-row-addr-type=0  --led-show-refresh --led-slowdown-gpio=2   --led-pwm-bits=9 --led-panel-type=FM6126A  --led-parallel=3 --led-pwm-dither-bits=2 --led-pwm-lsb-nanoseconds=100 -D0                                   283.2Hz  down to  221.9Hz

Both codebases show a start speed to 280Hz but then goes up to 330Hz on the old code but down to 220Hz on the new code.

old code is
c2076d4444ba70a93a55a627b63155ccf54277 (HEAD -> master)
Author: Henner Zeller <h.zeller@acm.org>
Date:   Mon Mar 16 18:38:21 2020 -0700

Before I start with a git bisect, did the speed computation get changed/fixed (I see the ouptut now adds "(lowest: 219.9Hz)" )?

hzeller, 2020-10-22:
Haven't looked at that recently, but I am not aware that the speed calculation would've changed too much. 
The 'lowest' values is now emitted to have a better starting value to be used for `--led-limit-refresh`.

So bisect might help finding the culprit.

marcmerlin, 2020-10-22:
will do

marcmerlin, 2020-10-22:
@gohlhausen  I see you contributed FM6127 support in https://github.com/hzeller/rpi-rgb-led-matrix/issues/964
Can you comment on what you think is going on with the 6127A init sequence and why it would only be going to one channel and not all 3?
Reading the code, I don't really see how the init is sent on all channels for FM6126A, but it is: https://github.com/hzeller/rpi-rgb-led-matrix/blob/b8c7cc42e7f5a852ba93c3cc70e4b31e576632c5/lib/framebuffer.cc#L471

marcmerlin, 2020-10-22:
Ok, this makes no sense, git bisect sent me to https://github.com/hzeller/rpi-rgb-led-matrix/commit/1362da5a2991a53f567c68c17f161731a66cdf8e which seems clearly innocent.

Still
```
root@rPi4:~/rpi-rgb-led-matrix# git log
commit 1362da5a2991a53f567c68c17f161731a66cdf8e (HEAD, refs/bisect/bad)
Author: Henner Zeller <h.zeller@acm.org>
Date:   Sat Aug 15 19:31:19 2020 -0700

    Simplify exit message (there is no timeout anymore).
root@rPi4:~/rpi-rgb-led-matrix# make; ./examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128  --led-row-addr-type=0  --led-show-refresh --led-slowdown-gpio=2   --led-pwm-bits=7 --led-panel-type=FM6126A  --led-parallel=3 --led-pwm-dither-bits=2 --led-pwm-lsb-nanoseconds=100 -D0 
make -C ./lib
make[1]: Entering directory '/root/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/root/rpi-rgb-led-matrix/lib'
make -C examples-api-use
make[1]: Entering directory '/root/rpi-rgb-led-matrix/examples-api-use'
make -C ../lib
make[2]: Entering directory '/root/rpi-rgb-led-matrix/lib'
make[2]: Leaving directory '/root/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/root/rpi-rgb-led-matrix/examples-api-use'
Size: 128x192. Hardware gpio mapping: regular
Press <CTRL-C> to exit and reset LEDs                                                             2 286.2HzReceived CTRL-C. Exiting.)z)z)

I get 280 to 320Hz. Then I go back one step and get back to 410Hz:

root@rPi4:~/rpi-rgb-led-matrix# git reset --hard HEAD^
HEAD is now at 54c63bf Update options of demo in README and help output.
root@rPi4:~/rpi-rgb-led-matrix# make; ./examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128  --led-row-addr-type=0  --led-show-refresh --led-slowdown-gpio=2   --led-pwm-bits=7 --led-panel-type=FM6126A  --led-parallel=3 --led-pwm-dither-bits=2 --led-pwm-lsb-nanoseconds=100 -D0 
make -C ./lib
make[1]: Entering directory '/root/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/root/rpi-rgb-led-matrix/lib'
make -C examples-api-use
make[1]: Entering directory '/root/rpi-rgb-led-matrix/examples-api-use'
g++ -I../include -Wall -O3 -g -Wextra -Wno-unused-parameter -c -o demo-main.o demo-main.cc
make -C ../lib
make[2]: Entering directory '/root/rpi-rgb-led-matrix/lib'
make[2]: Leaving directory '/root/rpi-rgb-led-matrix/lib'
g++ demo-main.o -o demo -L../lib -lrgbmatrix -lrt -lm -lpthread
make[1]: Leaving directory '/root/rpi-rgb-led-matrix/examples-api-use'                                          354.2HzSize: 128x192. Hardware gpio mapping: regular                                                                   373.7HzPress <CTRL-C> to exit and reset LEDs                                                             4 410.2HzReceived CTRL-C. Exiting.)z)z)

```

marcmerlin, 2020-10-22:
I've done a dist-upgrade on raspbian, gone from 4.19 to 5.4, gotten new compilers, rebooted, done make clean, and make, and I'm still seeing this.
I can't make any sense of it though.

Can anyone else reproduce?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin