hzeller/rpi-rgb-led-matrix #1183: AttributeError: 'rgbmatrix.core.RGBMatrixOptions' object has no attribute 'limit_refresh_rate_hz

louiselessel, 2020-10-24:
Hi, it looks like I am not gettin the newest version of the file for the python core when I run:

```
curl https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/rgb-matrix.sh >rgb-matrix.sh
sudo bash rgb-matrix.sh

```

I am putting this in my code:

```
# Configuration for the matrix

# - More info in ReadMe here https://github.com/hzeller/rpi-rgb-led-matrix
# - https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/rgbmatrix/core.pyx
options = RGBMatrixOptions()
options.rows = 32
options.cols = 32
options.chain_length = 11
options.parallel = 1
options.hardware_mapping = 'adafruit-hat'  # If you have an Adafruit HAT: 'adafruit-hat'
options.brightness = 100
options.pwm_bits = 11
options.pwm_lsb_nanoseconds = 300
#options.scan_mode = 0
#options.multiplexing =
#options.row_address_type =
#options.disable_hardware_pulsing =
options.show_refresh_rate = True
#options.inverse_colors = 
#options.led_rgb_sequence = 
#options.pixel_mapper_config = 
#options.panel_type = 
#options.pwm_dither_bits = 
options.limit_refresh_rate_hz = 60
options.gpio_slowdown = 2
#options.daemon = 
#options.drop_privileges = 

matrix = RGBMatrix(options = options)
```

**Error message:**
```
  File "./shader-on-matrix-All-Uniforms.py", line 134, in <module>
    options.limit_refresh_rate_hz = 60
AttributeError: 'rgbmatrix.core.RGBMatrixOptions' object has no attribute 'limit_refresh_rate_hz
```

If I look at this file on my computer it does indeed not contain limits_refresh_rate_hz.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/rgbmatrix/core.pyx

**Am I downloading the library wrong? Or is that file somehow not included right?**

daveythacher, 2020-10-25:
Found this:
```bash
# INSTALLER SCRIPT FOR ADAFRUIT RGB MATRIX BONNET OR HAT

# hzeller/rpi-rgb-led-matrix sees lots of active development!
# That's cool and all, BUT, to avoid tutorial breakage,
# we reference a specific commit (update this as needed):
GITUSER=https://github.com/hzeller
REPO=rpi-rgb-led-matrix
COMMIT=21410d2b0bac006b4a1661594926af347b3ce334
# Previously: COMMIT=e3dd56dcc0408862f39cccc47c1d9dea1b0fb2d2 
```

Which appears to be used here:
```bash
echo "Downloading RGB matrix software..."
curl -L $GITUSER/$REPO/archive/$COMMIT.zip -o $REPO-$COMMIT.zip
```

The age of that commit id is Oct 7, 2019. However the commit with that change is May 31, 2020.
You can simply try replacing the COMMIT id. So make this:
```bash
# hzeller/rpi-rgb-led-matrix sees lots of active development!
# That's cool and all, BUT, to avoid tutorial breakage,
# we reference a specific commit (update this as needed):
GITUSER=https://github.com/hzeller
REPO=rpi-rgb-led-matrix
COMMIT=21410d2b0bac006b4a1661594926af347b3ce334
# Previously: COMMIT=e3dd56dcc0408862f39cccc47c1d9dea1b0fb2d2 
```
Into this:
```bash
# hzeller/rpi-rgb-led-matrix sees lots of active development!
# That's cool and all, BUT, to avoid tutorial breakage,
# we reference a specific commit (update this as needed):
GITUSER=https://github.com/hzeller
REPO=rpi-rgb-led-matrix
COMMIT=6a0b9d145cf097907299e1a0b24aa6f3eb755dba
# Previously: COMMIT=21410d2b0bac006b4a1661594926af347b3ce334
```

You can do that manually with a text editor. However I have not tried this myself and do not know if there are other dependencies which are also outdated.

daveythacher, 2020-10-25:
This is kind of terrible. However this project handles the Python a little strange. Usually you have a release branch which is stable and you can track that. Technically the python should be its own repository which tracks this repository via submodule against specific commit id. That way its release branch is stable till updates come to bring it up to speed.

That script attempts to download, configure and build the library for you. It appears to support Adafruit products which makes sense. It does stream line things nicely. However it is 9 months old and uses manual commits. Probably to avoid broken master, which I am not sure I can fault them for.

louiselessel, 2020-10-26:
I made my own copy of the rgb-matrix.sh file from https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/rgb-matrix.sh and replaced the lines as you said. Put it on the desktop of my pi. 
Then ran 'sudo bash rgb-matrix.sh', which worked, and I then manually had to put the 'rpi-rgb-led-matrix' file folder in the 'pi' folder.  I now have access to all of the options. Thank you!

And yes, maybe Adafruit should update their script :)

closed 2020-10-26 by louiselessel