hzeller/rpi-rgb-led-matrix #1185: Displaying graphics outputs from applications like Processing or Pd GEM

fhchl, 2020-10-26:
Thanks for this cool library!

How would one display real time graphics from applications like [Processing](https://www.processing.org/) or the [GEM library of Pure Data](https://gem.iem.at/) on a LED matrix? Can this library be used for that purpose? 

Reading the manual, I can only find examples on how to directly programm the LED, while I would rather like the tools above to generate the graphics output and just display that on the panels. Any ideas?

I found https://github.com/adafruit/rpi-fb-matrix which sadly does not support my Pi4.

daveythacher, 2020-10-26:
This is off topic. Honestly I think you should look into the code to understand more. You clearly have an idea of what you are after. I think you made this clear too. To answer this question, no. This library does not really get involved with that. Many of the library surrounding this library are not overly optimized or structured correctly.

Overall this library pushes some limits uncomfortably. This will begin aggravate this. It can be done in certain cases and applications. To understand that I would recommend understanding the code and how the Pi works. I know there is a few benefits to doing this. It effectively uses the GPU to draw the frames rather than the CPU.

I am curious to know what kind of setup and application that rpi-fb-matrix targets. Along with performance and stability numbers. I know there are LED drivers which will be more conductive to this. This project however does not support them. It would also be pretty much a rewrite to structure it into a format capable of supporting them. This library is very much influenced by one driver type.

This library just abstracts that type and presents it as entirety of LED matrixes. There are good reasons for why the different LED drivers were created. A code base could in theory abstract any type. However this one struggles with abstracting ones very similar to it like FM612x drivers. This is due to a difference in convention that exists within the LED driver. This convention is standard to the other types for the most part.

The way these drivers work they can have issues with memory, cache, processor, etc. performance. This library can do some of this, if you pay the correct respect to some of the moving parts. Unfortunately, I found this will be up to you. There is little documentation, knowledge from issue tracker, and many contributors are very unhelpful. There is several things you need to be on the look out for. PWM bit count, OE time, column length, peak power, GPIO bit rate, cache/memory stability, overhead, etc. 

For low cost applications that are not demanding in performance or overly large this library will work. However I would not recommend it for commercial applications without serious detailed thought. It really is a shame it does not support FM612x as these would allow control of peak power. You can mange all of these with software following this approach, however to solve them you need to move to other methods. I tried to explain this many times.

fhchl, 2020-11-02:
Thanks for your thoughts!

> I am curious to know what kind of setup and application that rpi-fb-matrix targets.

The repo is advertised [in this article from Adafruit](https://learn.adafruit.com/raspberry-pi-led-matrix-display). I guess an obvious application is to just have fun and play around with the display without programming it directly. 

> This library is very much influenced by one driver type.

Can you recommend any resources on reading up on LED drivers? I don't have any experience with it yet.

daveythacher, 2020-11-03:
This project is mostly influenced by serial shift registers with constant current sink on outputs. This going to be a long, I hope this is useful.

These are likely your standard shift registers but they have a few adaptions that make them better suited for LED matrixes. Most notably the current limiters which will respond to the LED voltage to control the current moving through the LEDs. Note the extra voltage is dropped on the driver which moves to heat. (Like a linear regulator does.) This will waste energy and can create thermal problems for driver and PCB. 

These panels are designed to work at 5V however the drivers may support higher voltage but may need layout consideration adjustments. Also consult datasheet for max thermal, power, voltage, etc. implications. Note the power rail is shared so check all components if you ever plan to attempt to drive above 5V. Many support going below 5V however the current limiters are set for white color operation at 5V. 

Lowering this will require some adjustment. Certain LED colors which operate at lower voltage will be at max current while the higher voltage colors will be at lower current. Look up IV plot for diodes to understand more. LEDs are basically just diodes so this applies to them, and is why the constant current generators are required. Note many LED will work at high burst current but this requires very careful control. Which may not be recommended for multiplexed displays.

Other drivers have dedicated PWM generators on the outputs which reduces the traffic by (2 ^ (pwm_bits - log2(pwm_bit))) times. This can be significant, for 8 bits it is a reduction of 32 times and for 16 it is a reduction of 4096 times. This allows for longer chains or lower serial clocks.

Then there is memory mapped LED drivers which hold a small section of the LED matrix internally. This can reduce the traffic further by multiplex ratio times refresh rate. So if a panel is 1:8 and has 95Hz refresh rate this would be a reduction of 760 times. Which again allows for long chains or lower serial clocks. Or for the LED panel high speed burst transmission, which gives it some wiggle room against timing violations by less than ideal real time system.

Some drivers support current control via software. FM612x appear to be standard shift register ones with this support. However i have never found any datasheet which explains where those are enabled in the configuration registers. However the datasheets to mention them if you translate them. Many memory mapped LED drivers support this.

Software current gain control can limit peak power consumption. Which can be used to avoid issues with power supplies or batteries. If not properly handled the power supply may shutdown or could suffer damage. You could hard set this with a resistor. However there are cases where you may want to change this on the fly. POE/POE+ and LiPo batteries with switch mode regulators come to mind.

I should mention that there is a catch with the PWM drivers. They usually require and external clock which drives PWM counters and state machine logic. This clock source needs to be fast for certain refresh rates, multiplex ratios, and pwm bit counts. This will cap the actual FPS of the display.

The reason for using high pwm bit counts is to smooth out the IV plot of LEDs, which if familiar with ideal diode equation is exponential. One way to smooth this out is to use high bit count against low bit count. Express 8 bits of linear information using 16 bits of log information.

Attempting to drive high PWM bit counts over a decent size serial shift register panel will fail due to bandwidth along. As I mentioned earlier. Now BCM which this library uses will help with the CPU load. PWM bit generation is very IO centric, and software is terrible at IO. Software was originally due to manufacturing issues of complexity and size. 

Software has always been better for tying things together. Primary compute using software was reserved for extreme circumstances. However not all software follows assembly/processor model. FPGAs for example support a much more direct model which allows them to be flexible while not wrestle with the same traditional IO limitations.

However BCM uses the properties of binary representation and timing to express the same information without repeating work. It will also get a processing power reduction of (2 ^ (pwm_bits - log2(pwm_bits)). However the bandwidth requirements still remain. Timing is important for this case.

Other approaches for improving computation also exist. Such as wasting memory to lower computational requirements. However bandwidth of computation must be persevered for worst case according to max FPS. IO offloading can work for this model, such as interrupts and DMA. (Assuming supported by sub system implementation. Which is an issue for Pi.)

LED Matrixes are effectively memories, which is why the look like ROMs. Multiplexing is handled like addressing a ROM. It uses much of the same decoder logic. You have IO, digital logic, software, architecture, parallelization, timing, etc. A lot of concepts that are at the core of computing have parallels inside LED Matrixes. So simple yet many of the basic concepts are there. Not much has really changed.

Now this library mostly supports the first type that I mentioned. They appear to be more common. They are easier to work with. However they have a few properties which are annoying for direct GPIO drive on Pi. They have size lim [... cut ...]  I will not get into that as my knowledge of that is rudimentary. 

Lot of PWM drivers use a modified type of PWM called scrambled PWM. It goes by a few other names too, depending on vendor. 

For more information about LED matrixes, I would recommend google. A lot of people have built LED matrixes and there is code and tutorials for most of the basics. Note LED cubes are really LED matrixes.

For information about LED drivers, I recommend looking up datasheets. ST, TI, Macroblock, ChipOne, etc.

daveythacher, 2020-11-03:
Note LED sender cards can receive input from video inputs and convert them to LED receiver cards which drive LED panels. Then all you do is generate a monitor output. No programming required. Cost can vary. So if that is what you are looking for I am not sure why you would use a Pi.

I do not know if Adafruit has anything like this still:
https://www.adafruit.com/product/1453

daveythacher, 2020-11-03:
There are a few caveats to the numbers that I mentioned depending on certain applications. For example the memory mapped gain assumes static image not video. The PWM driver assumes the needed bit count is the actual bit count of the driver. However these do not massively impact the point. These conditions also can be true depending on application and LED driver in question.

daveythacher, 2020-11-03:
Note this library does attempt to optimize a few things. Such as using a dedicated core. However there is only so much that can be done about bandwidth and entropy. These issues even with the Pi can be worked around some too. Some may question the point of this to some extent. 

Now for entertainment or educational purposes this can very useful. Or if making something for specific usages which are harder to accomplish due to existing outside expected usages of existing alternatives. I can think of a few cases that may fit that category.

davemaster, 2020-11-03:
> Thanks for this cool library!
>
> How would one display real time graphics from applications like [Processing](https://www.processing.org/) or the [GEM library of Pure Data](https://gem.iem.at/) on a [...]

It  supports RPi 4, but something changes in rpi-fb-matrix, and needs some update to work with new rpi-rgb-led-matrix version.

fhchl, 2020-11-05:
Really? When I tried the repo out, the Pi stayed black for me. Can you elaborate on what you mean with "but something changes in rpi-fb-matrix"?

fhchl, 2021-01-05:
For whoever would like to do the same thing. I have written a small external that is a wrapper around this library that makes it easy to control LEDs with the Raspberry Pi: https://github.com/fhchl/ltbl

closed 2021-01-05 by fhchl

hzeller, 2021-01-05:
Note, there is also https://github.com/hzeller/rpi-matrix-pixelpusher which implements the PixelPusher protocol, which works with Processing.