hzeller/rpi-rgb-led-matrix #1186: Can I sync separate displays into one?

rijotech, 2020-10-26:
I was wondering if there is a way to synchronize multiple panel into one, meaning let's say 4 separate raspberry pi connected via lan or wifi running a set of 384x192 modules into one big panel and each one displaying a part of a video or image?

daveythacher, 2020-10-27:
Yes and no. You can do this but it would be a little insane. With FPGA or hardware you can easily. Linux overall works best as master. However if the underlying hardware supported this it could work. It could master DMA configurations which would internally become slaves. 

Event systems and timing does not like being slaves. This can become blocking. If you use parallelization via hardware you can avoid all of this. Multicore processors can be very effective by emulating this, however they can get tied up in the memory/cache subsystem complications playing one size fits all games. However that gets into a tone of off topic stuff. 

Note if you have a lot of pins you can also virtualize different panel notions together. Really this is kind of what you are looking to get, besides more cores. This is a load which does not really need UMA for certain sections. Syncing however in this case will represent inter-process communication which can result in bottlenecks. IO bottleneck can cripple performance. UMA can relieve some of this internally. NUMA simulates IO bottleneck just like Ethernet.

Note this assumes you are trying to sync them in terms of row lines. Basically attempting to drive a ton of data pins on the same clock edge. For example have a SPI master generate clock for multiple SPI slaves where the data lines come from the slaves.

Now if you are attempting to simply partition the panel in terms of data/update/drive locality you can do that. Ethernet would do that easily. Have a master send out packets to multiple Pi's which drive sub sections of the display. This would be a version of virtualizing different panel notions together also. Abstraction happens via Pi from protocol data to actual serial data.

This would be a form of parallelization. Now you would have to consider certain bandwidth notions, but likely the Pi will cap out before Ethernet will. Many of the issues facing compute/processor systems in terms of performance/time is the generic serial nature.

However this parallelization is not synchronous but rather asynchronous. The server has main image which it carves up to clients. All of this could be Pi's. You could PXE boot the clients if you wanted, but they will need ID number. Again looking more like NUMA. Lot of hell in computing comes from syncing in UMA. If willing to let that go, you can just do NUMA.

daveythacher, 2020-10-27:
The big panel walls are usually tied together via Ethernet switch (full-crossbar). The wall is divided into modules each of which has a controller module inside of it. This controller module usually uses Ethernet. It is called a receiver card. These all connect back to a sender card where the main display signal is presented. Together they create a huge display like a TV or monitor.

They have other functions and this is simplified but I think it is good enough for this.

daveythacher, 2020-10-27:
That or they use daisy-chain. Which is another option.

rijotech, 2020-10-27:
Wow thanks for the info I knew some of it I was just wondering if I could just perform a cluster and using rsync and pssh to control all the computers from one but that may be a lot of work, I will try to develop a method if there is a way I may find out.

daveythacher, 2020-10-27:
What you can do is PXE boot. This will require a DHCP server and TFTP server as I recall. You can get ID chips which you can use to create MAC addresses. Basically the kernel modules will load up, check the ID number, generate MAC Address and make DHCP request once OS is loaded. DHCP server will have reserved IP address for those MACs.

What this gives you is a single boot image for all clients. You then know the IP address which correspond with those sections. You can create sockets against those IP addresses. Then transport data to them.

You can build a multicast tool which will send request for MAC address or ID. Then you can figure out which MAC address or ID maps to the section. So once you have that boot image you never touch it again. The hardware automatically creates the difference for you. This will not change, so you only have to do discovery once.

This PXE image would be beyond this scope of this project. 

However I have toyed around with modifying a kernel module(s) to read a Microchip EEPROM chip with a 48 bit MAC address and passing this to the kernel module for an ENC28J60. Many DHCP servers exist in routers which may support reserving IPs. If not a Linux DHCP server likely exists which can installed on a Pi. Linux TFTP servers also exist.

Now you could make a user space script which does most of that kernel module stuff. You could have it run at boot. The install and configuration of the servers should be straight forward. Generating the PXE boot client on the Pi, is the part I am not sure of. however I think others have done this.

Next comes writing the server and client program. Which I am not sure what you have in mind for that. Then there is the multicast program which could be straight forward.

Actually you could drop PXE and just clone the SD cards. They would use the same script and client programs. No need for TFTP server. You could also hard code the IP address which would allow you to drop the ID chips and DHCP reserving. This would put you with a just a server and client program. With fixed IP you could use RSYNC and PSSH.

All of this is beyond the scope of this project.

daveythacher, 2020-10-27:
Here is several alternatives to an ID chip:
https://raspberrypi.stackexchange.com/questions/42145/raspberry-pi-hardware-id

It mentions the Pi's internal serial number, USB Ethernet MAC address, etc.

jaygarcia, 2020-11-24:
Are you looking to do something like this? https://www.youtube.com/watch?v=s0lbVMcgfww
(broken up prototype) https://www.youtube.com/watch?v=gTtBLEOPpsM&feature=youtu.be

I've done it but my implementation isn't fully documented and I concur with @daveythacher that there are better ways and even cheaper alternatives. 

However if you're interested in at least learning what we did, here's documentation on how we did it (inside our rather incomplete library). It's my first attempt at LibASIO and even TCP/IP programming, so it's likely less than optimal -- but it works. :D 

https://github.com/ModusCreateOrg/network-rgb-matrix-display/blob/master/md/How_it_works.md

rijotech, 2020-12-09:
I will definitely find a way to this thanks you @jaygarcia

closed 2020-12-09 by rijotech

rijotech, 2020-12-17:
Mr @jaygarcia after playing around for a few days I realized that it will take me a really long time to figure it out, and I was wondering if there is a way you could provide me some basic information related to how to make 
network-rgb-matrix-display work. I able to compile the server `matrix-server` and client `network-display-client` I kind of understand a bit but I'm confused, can your please provide me a few example on how using this to do `one client, many servers` unlike most people here I'm self-taught and it can be kind of hard to work without some sort of documentation.
Thank You in advance!