hzeller/rpi-rgb-led-matrix #1189: Sryled 64x64 + Adafruit Bonnet

FredericoCoelhoNunes, 2020-11-01:
## Description

I've been trying to play around with a 64x64 matrix (bought in on Alibaba from the Sryled providers), but am running into some issues.

Everything seems to be working except that there are two large black stripes in the matrix that don't light up, alternating with two strips which are working.

## Questions

I am almost sure that this problem is due to the fact that the E address line is not connected. This user had a similar problem here (https://github.com/hzeller/rpi-rgb-led-matrix/issues/961) and fixed it by simply soldering the E line to the 8 on the back of the bonnet (if I'm not mistaken, it's simply heating up a small blob of metal and dragging it to the 8).

However, I also found two other sources of information:

* in this Adafruit tutorial, they do one extra step, and solder pin 4 to pin 18: https://learn.adafruit.com/matrix-led-sand/assemble
* I have also read in several sources that for some 64x64 matrices, I am supposed to connect the E line to the 16, instead of the 8 - however I can't find any documentation on how to discover which option is correct.

Furthermore, I have also found this section on the README of this repository (https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet) which makes it look even more complicated, although I think it's referring to an old version of the bonnet.

I would really appreciate some guidance on how to proceed, given that I have very basic knowledge of electronics and would hate to fry my hardware!

Thanks a lot for reading.

lornajane, 2020-11-20:
Does it look like this?
[image]

This is a 64x64 panel with an Adafruit Hat and it failed like the picture shown. It's working perfectly now when I use `--led-gpio-mapping=adafruit-hat` when running the examples. Maybe that could help you too?

FredericoCoelhoNunes, 2020-11-23:
Hi @lornajane ,

doesn't look quite like that! The illuminated strips are larger, there's only two of them, about the same width as the black strips.

Thanks!

FredericoCoelhoNunes, 2021-01-03:
We've tried soldering a certain pad to 8 and then to 16, as suggested here: https://learn.adafruit.com/adafruit-rgb-matrix-bonnet-for-raspberry-pi/driving-matrices (on Step 3), but to no avail. Even bought a soldering iron specifically for this haha.

Still seeing two large stripes of black, alternating with two fully colored stripes. Any advice would be very welcome!

hzeller, 2021-01-03:
Did you [follow the instructions](https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet) in the readme to add the E line ?

mariana-s-fernandes, 2021-01-16:
Hi, I was working with @FredericoCoelhoNunes on this project and I finally discover how to make this work. Unfortunately the instructions are not complete for the [Adafruit bonnet we are using](https://learn.adafruit.com/adafruit-rgb-matrix-bonnet-for-raspberry-pi). I leave here a for "dummies" (like me) procedure for someone that may be facing this same issue.

1st step: On the back on my matrix, I found out the HUB75 pinout mapping and, comparing with the mapping on the bonnet, I understood that I needed to connect E line to pin 8, as described in the instructions. So I soldered them together.

2nd step: I tried it and it was still not working, and after some exhaustive Google search I found out that I also needed to wire pins 4 and 18 together. Here is the [page](https://learn.adafruit.com/motion-controlled-matrix-bed-clock/assembly) where I found that out. So I tested by connecting a wire (didn't even solder it yet) between those pins.

3rd step: When I ran the command again, the stripes had changed - the ones that were previously blank had image and vice-versa. Then I tried to play around with the flags and I found out I needed to change to `--led-multiplexing=0`. And it finally worked!!

Here is the full command that made it work in the end:
`sudo ./demo -D0 --led-cols=64 --led-rows=64 --led-gpio-mapping=adafruit-hat --led-multiplexing=0 --led-show-refresh --led-slowdown-gpio=4`.

Hope this helps someone :)

FlippinBerger, 2021-11-18:
@mariana-s-fernandes this is exactly what we were looking for! Took like 20 minutes to get the extra connections soldered together and now our Rust code is working perfectly for targeting the matrix. Thank you!

closed 2025-11-27 by marcmerlin