hzeller/rpi-rgb-led-matrix #1190: 16x16 1:1 static outdoor RGB matrix

craiglee22, 2020-11-04:
Dear Sir,
We'd like to use Full Color RGB module 10,000cd/m2 for outdoor. I had been searched your lib and issues, however it did not find right resolution.
16x16 module with 3 chains such as,
$ sudo ./demo -D0 --led-cols=16 --led-rows=16 --led-chain=3 --led-gpio-mapping=regular --led-no-hardware-pulse --led-slowdown-gpio=4 

We're using HUB75, Raspberry Pi 4B and it had no ABCD address lines and those pins are GND everything same except address pins. HUB75 last pin GND is CLK, so two CLK(no meaning I think, pin 13 CLK use)
It is 1:1 or 1/1 or static RGB full graphic module.
The direct pixel is drawing by attached sequence, how to implement or use your library.
The fulls are okay, however drawing fonts and cube are not working properly. Is there any comment using your library rpi-rgb-led-matrix-master.

[image]
[image]

daveythacher, 2020-11-04:
Interesting, I considered this many years ago. It basically is a memory mapped using shift registers. If this is the case you will need to use mapper. I think that will allow you to convert the pixels locations.

daveythacher, 2020-11-04:
I am surprised as this is fairly expensive to do it this way. However it could be simpler to some extent. You can also multiplex with a raw number of drivers if you can tri-state the outputs. PWM LED drivers may play nice with this. However they may not play nice with this library and the way they are usually power here.

However what you want to do is implement a pixel-mapper if I am not mistaken. Here is a few examples:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/pixel-mapper.cc

daveythacher, 2020-11-04:
I know why they did this. It is likely very bright and uses a decent amount of power too. Does it use two chains: R0, G0, B0 and R1, G1, B1?

daveythacher, 2020-11-04:
You may need a custom pin mapping. Basically comment out the address lines, so they do not get driven against ground. I think you will want to tell it to use one row, but I am not sure what this does exactly. 

If you want you can set the rows to 16, but you will need to make all the rows the same data. This would be less ideal compared to setting the rows to 1. I think that may work, but I am not sure. You can draw frames that way with this kind of panel, but I would not quote me on that either.

You have performance concern purely against columns not rows. So what you will want to do is put down one row with 128 columns, which is the same number of pixels as 16x16 assuming 1:8 multiplex using two serial chains.

craiglee22, 2020-11-04:
Yes two chains (R1,G1,B1),(R2,G2,B2),OE,LAT,CLK and when I tried to draw SetPixel(x,y,r,g,b), then it displayed 8 different location as attached. It implemented U-mapper;Rotate=180 two months ago for another project and it worked well 1/2 scan. This is 1/1 new LED module. I'm not sure it changes pixel map based on CLK.

[image]

craiglee22, 2020-11-04:
This is 16x16 one chain to debug rather than 16x16 3 chain, so 48x16 and it is same result.

daveythacher, 2020-11-04:
How many LED drivers are on the back? So is a pixel repeated in 1 out of 16 locations or just 1 out of 128?

daveythacher, 2020-11-04:
--led-rows=16 may explain some of this. Try --led-rows=2 if you think this is 1:1 using two RGB chains.

craiglee22, 2020-11-04:
When I tried to find out, it had big frames stencil. It only can see capacitors also need to disassemble fully.

craiglee22, 2020-11-04:
It is not allowed --led-rows=2 and displays warning as --led-rows=16 or 32 on terminal window.

craiglee22, 2020-11-04:
The brand is different, however product is same 240mmx240mm P15 16x16 pixel module.

https://streetcommunication.com/wp-content/uploads/2020/02/Tuforce-technical-specification.pdf

[image]
[image]
[image]
[image]

daveythacher, 2020-11-04:
Currently one pixel shows up in 8 locations of panel and you believe the panel is 1:1 using two RGB shift register chains? To test this you would just see if you can address the pixel uniquely. To do this --led-cols=128 with --leds-rows=16. Which you may have done. 

You are working with one of the sub panels that exists within these cases? Are you attempting to replace the controller module with a Pi? Usually there is a receiver card inside these cases along with power that talks to a sender card.

Now assuming that works and you just duplication row data you should get the entire image expressed in the columns according to the mapping you find. After this a pixel mapper can be used to convert the 1x128 over two RGB chains to 16x16 over two RGB chains. 

A custom pin mapper can be created to remove the unneeded address pins.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/b8c7cc42e7f5a852ba93c3cc70e4b31e576632c5/lib/hardware-mapping.c#L25-L68
Change that to this:
```C++
  /*
   * The regular hardware mapping described in the wiring.md and used
   * by the adapter PCBs.
   */
  {
    .name          = "regular",

    .output_enable = GPIO_BIT(18),
    .clock         = GPIO_BIT(17),
    .strobe        = GPIO_BIT(4),

    /* Address lines */
    //.a             = GPIO_BIT(22),
    //.b             = GPIO_BIT(23),
    //.c             = GPIO_BIT(24),
    //.d             = GPIO_BIT(25),
    //.e             = GPIO_BIT(15),  /* RxD kept free unless 1:64 */

    /* Parallel chain 0, RGB for both sub-panels */
    .p0_r1         = GPIO_BIT(11),  /* masks: SPI0_SCKL  */
    .p0_g1         = GPIO_BIT(27),  /* Not on RPi1, Rev1; use "regular-pi1" instead */
    .p0_b1         = GPIO_BIT(7),   /* masks: SPI0_CE1   */
    .p0_r2         = GPIO_BIT(8),   /* masks: SPI0_CE0   */
    .p0_g2         = GPIO_BIT(9),   /* masks: SPI0_MISO  */
    .p0_b2         = GPIO_BIT(10),  /* masks: SPI0_MOSI  */

    /* All the following are only available with 40 GPIP pins, on A+/B+/Pi2,3 */
    /* Chain 1 */
    .p1_r1         = GPIO_BIT(12),
    .p1_g1         = GPIO_BIT(5),
    .p1_b1         = GPIO_BIT(6),
    .p1_r2         = GPIO_BIT(19),
    .p1_g2         = GPIO_BIT(13),
    .p1_b2         = GPIO_BIT(20),

    /* Chain 2 */
    .p2_r1         = GPIO_BIT(14), /* masks TxD when parallel=3 */
    .p2_g1         = GPIO_BIT(2),  /* masks SCL when parallel=3 */
    .p2_b1         = GPIO_BIT(3),  /* masks SDA when parallel=3 */
    .p2_r2         = GPIO_BIT(26),
    .p2_g2         = GPIO_BIT(16),
    .p2_b2         = GPIO_BIT(21),
  },
```

Add a null RowAddressSetter to lib/framebuffer.cc
```C++
class NullRowAddressSetter : public RowAddressSetter {
public:
  NullRowAddressSetter (int, const HardwareMapping &) {}
  virtual gpio_bits_t need_bits() const { return 0; }
  virtual void SetRowAddress(GPIO *, int) {}
};
```
https://github.com/hzeller/rpi-rgb-led-matrix/blob/b8c7cc42e7f5a852ba93c3cc70e4b31e576632c5/lib/framebuffer.cc#L439-L441
Change that to this:
```C++
case 4:
    row_setter_ = new SM5266RowAddressSetter(double_rows, h);
    break;
case 5:
    row_setter_ = new NullRowAddressSetter (double_rows, h);
    break;
```
https://github.com/hzeller/rpi-rgb-led-matrix/blob/b8c7cc42e7f5a852ba93c3cc70e4b31e576632c5/lib/options-initialize.cc#L392-L395
Change that to this:
```C++
if (row_address_type < 0 || row_address_type > 5) {
    err->append("Row address type values can be 0 (default), 1 (AB addressing), 2 (direct row select), 3 (ABC address), 4 (ABC Shift + DE direct), 5 (NULL).\n");
    success = false;
  }
```

Finally a code modification for allowing multiplex ratios down to 2.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/b8c7cc42e7f5a852ba93c3cc70e4b31e576632c5/lib/options-initialize.cc#L367-L371
Change that to this:
```C++
if (rows < 2 || rows > 64 || rows % 2 != 0) {
    err->append("Invalid number or rows per panel (--led-rows). "
                "Should be in range of [2..64] and divisible by 2.\n");
    success = false;
  }
```
This technically is not correct due to this:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/b8c7cc42e7f5a852ba93c3cc70e4b31e576632c5/lib/framebuffer.cc#L40-L44

daveythacher, 2020-11-04:
Don't quote me on this but this should convert the pixel locations to actual locations. It make 0,0 the upper left, which I am not sure is correct or not.

```C++
void MapVisibleToMatrix(int matrix_width, int matrix_height, int visible_x, int visible_y, int *matrix_x, int *matrix_y)
     *matrix_x = visible_x / 8;
     *matrix_y = (((visible_x % 8) == 0) * 64) + ((visible_y / 4) * 16) + ((visible_x % 4) * 4) + (visible_y % 4);
}
```

merliongold, 2020-11-05:
I'm debugging due to not working properly and I'll let you know.
$ sudo ./minimal-example --led-gpio-mapping=regular --led-rows=16 --led-cols=128 --led-chain=1 --led-row-addr-type=5 --led-no-hardware-pulse --led-slowdown-gpio=0 --led-multiplexing=1 -f ../fonts/10x20.bdf A

$ sudo ./minimal-example --led-gpio-mapping=regular --led-rows=2 --led-cols=128 --led-chain=1 --led-row-addr-type=5 --led-no-hardware-pulse --led-slowdown-gpio=0 --led-multiplexing=1 -f ../fonts/10x20.bdf A
minimal-example: framebuffer.cc:314: rgb_matrix::internal::Framebuffer::Framebuffer(int, int, int, int, const char*, bool, rgb_matrix::internal::PixelDesignatorMap**): Assertion `rows_ >=4 && rows_ <= 64 && rows_ % 2 == 0' failed.
Aborted

It also needs to modify when it uses --led-rows=2
void MapVisibleToMatrix will be added in lib/multiplex-mappers.cc and is that correct?

class MultiplexMapperBase : public MultiplexMapper {
public:
  ```
  virtual void MapVisibleToMatrix(int matrix_width, int matrix_height,
                                  int visible_x, int visible_y,
                                  int *matrix_x, int *matrix_y) const {

 // const int within_panel_x = visible_x % panel_cols_;
 // const int within_panel_y = visible_y % panel_rows_;

 // MapSinglePanel(within_panel_x, within_panel_y, &new_x, &new_y);
    *matrix_x = visible_x / 8;
    *matrix_y = (((visible_x % 8) == 0) * 64) + ((visible_y / 4) * 16) + ((visible_x % 4) * 4) + (visible_y % 4);
  }
  ```
};

daveythacher, 2020-11-05:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/b8c7cc42e7f5a852ba93c3cc70e4b31e576632c5/lib/framebuffer.cc#L302
Change that to
```C++
assert(rows_ >=2 && rows_ <= 64 && rows_ % 2 == 0);
```

daveythacher, 2020-11-05:
You are modifying a multiplex-mapper. I was suggesting a pixel-mapper. You can reimplement on of the existing ones for testing if you want. Here is the rotate logic: https://github.com/hzeller/rpi-rgb-led-matrix/blob/b8c7cc42e7f5a852ba93c3cc70e4b31e576632c5/lib/pixel-mapper.cc#L66-L87

My previous comment will remove the assert error from framebuffer.cc

merliongold, 2020-11-05:
Please refer to below and it is not working and it needs to figure it out about the (x, y) to (new_x, new_y).

class RotatePixelMapper : public PixelMapper {
public:
  RotatePixelMapper() : angle_(0) {}

  ```
  virtual void MapVisibleToMatrix(int matrix_width, int matrix_height,
                                  int x, int y,
                                  int *matrix_x, int *matrix_y) const {
    int visible_x = x;
    int visible_y = y;
    switch (angle_) {
    case 0:
      *matrix_x = visible_x / 8;
      *matrix_y = (((visible_x % 8) == 0) * 64) + ((visible_y / 4) * 16) + ((visible_x % 4) * 4) + (visible_y % 4);
      break;
    case 90:
      *matrix_x = matrix_width - y - 1;
      *matrix_y = x;
      break;
```
};

$ sudo ./minimal-example --led-gpio-mapping=regular --led-rows=2 --led-cols=128 --led-chain=1 --led-row-addr-type=5 --led-no-hardware-pulse --led-slowdown-gpio=0 --led-multiplexing=4 --led-pixel-mapper=Rotate:0
minimal-example: framebuffer.cc:315: rgb_matrix::internal::Framebuffer::Framebuffer(int, int, int, int, const char*, bool, rgb_matrix::internal::PixelDesignatorMap**): Assertion `rows_ >=4 && rows_ <= 64 && rows_ % 2 == 0' failed.
Aborted

$ sudo ./minimal-example --led-gpio-mapping=regular --led-rows=16 --led-cols=128 --led-chain=1 --led-row-addr-type=5 --led-no-hardware-pulse --led-slowdown-gpio=0 --led-multiplexing=4 --led-pixel-mapper=Rotate:0

after changing in lib/framebuffer.cc  like this and rows_ value is called this function to 1  ,
#ifdef  ORG_SRC
  assert(rows_ >=4 && rows_ <= 64 && rows_ % 2 == 0);
#else
  printf("FrameBuffer::Framebuffer>rows_ %3d \n", rows_);
  assert(rows_ >=1 && rows_ <= 64 && rows_ % 2 == 1);
#endif

$ sudo ./minimal-example --led-gpio-mapping=regular --led-rows=2 --led-cols=128 --led-chain=1 --led-row-addr-type=5 --led-no-hardware-pulse --led-slowdown-gpio=0 --led-multiplexing=4 --led-pixel-mapper=Rotate:0
FrameBuffer::Framebuffer>rows_   1
Floating point exception

daveythacher, 2020-11-05:
> assert(rows_ >=1 && rows_ <= 64 && rows_ % 2 == 1);
Missed that.

What changes have you made? Any chance you want to commit to a feature branch?

daveythacher, 2020-11-05:
Just comment out that assert.

craiglee22, 2020-11-05:
I'm going to remapping PixelMapper accordingly width x height due to out of range using
--led-cols=128 --led-rows=2
--led-cols=64 --led-rows=4

[image]

craiglee22, 2020-11-05:
> Just comment out that assert.

It had floating point exception above statement.

craiglee22, 2020-11-05:
I think it is displaying two places with bypass PixelMapper x,y same rather than eight using
--led-cols=64 --led-rows=4 --led-row-addr-type=5 --led-pixel-mapper=Rotate:0

It should be fixed to work properly without floating exception, 
--led-cols=128 --led-rows=2 --led-row-addr-type=5 --led-pixel-mapper=Rotate:0

merliongold, 2020-11-05:
It works SetPixel(x,y) as one pixel as following command and I'll modify MapVisibleToMatrix to explicit x,y rather than bypass x, y accordingly,

$ sudo ./minimal-example --led-gpio-mapping=regular --led-rows=2 --led-cols=128 --led-chain=1 --led-row-addr-type=5 --led-no-hardware-pulse --led-slowdown-gpio=4 --led-multiplexing=0 --led-pixel-mapper=Rotate:0

daveythacher, 2020-11-05:
Is this a school project or something?

craiglee22, 2020-11-05:
It should be pointed 000,0(physically 0,4 as picture)to real 0, 0 position after changing MapVisibleToMatrix using lookup table or calculation mathematical I think.
[image]

craiglee22, 2020-11-05:
> Is this a school project or something?

It is the sports stadium display.

craiglee22, 2020-11-05:
Thank you for your supporting. It implemented well using lookup table based on drawing direction w/o ABCD address and it is displayed (0,0) to (384-1,1) accordingly as attached snapshots using minimal-example which changes SetPixel(0,0) to SetPixel(383, 1) loop for 16x16 3pcs(48x16) aka 384x2. However, it is not displayed ./demo or ./scrolling-text-example.

$ sudo ./demo -D0 --led-cols=128 --led-rows=2 --led-no-hardware-pulse --led-chain=3 --led-slowdown-gpio=4 --led-pixel-mapper=Rotate:0 --led-multiplexing=0 --led-row-address-type=5 --led-gpio-mapping=regular

craiglee22, 2020-11-05:
Is it okay to display and cube/scrolling text on led-rows=2?

The following SetPixel are working good 0,0 to 383,1.
[image]
[image]

daveythacher, 2020-11-06:
I was under the impression that setPixel would work on 16x16 grid locations. The mapper would convert the locations from 16x16 to 128x2, via MapVisibleToMatrix. Demo would work using 16x16, however the underlying panel would work 128x2. On command line you would specify 128x2; however the demo I think will see 16x16, as the API will report back 16x16. 

Application logic is consistent with actual display appearance while library is consistent with panel implementation. This is my understanding of it perhaps I am mistaken. This is getting a little outside my understanding of the intention/recommend usage.

I forgot to mention this:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/b8c7cc42e7f5a852ba93c3cc70e4b31e576632c5/lib/pixel-mapper.cc#L53-L64

Here is the function that uses these methods:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/b8c7cc42e7f5a852ba93c3cc70e4b31e576632c5/lib/led-matrix.cc#L550-L580

You could change to:
```C++
virtual bool GetSizeMapping(int matrix_width, int matrix_height, int *visible_width, int *visible_height) const { 
     *visible_width = matrix_width / 8;
     *visible_height = matrix_height * 8;
     return true; 
 } 
```

Again with a grain of salt.

craiglee22, 2020-11-06:
Thank you so much and it works excellently. 😃😊😆

closed 2020-11-06 by craiglee22