hzeller/rpi-rgb-led-matrix #1191: Looking for a 64x32 or 64x64 matrix

ISaneI, 2020-11-05:
Hey I'm currently searching for a 64x32 matrix panel on Aliexpress

This is the [ Matrix Panel Drive Board](https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-for-raspberry-pi-v2/) I want to buy.

But now I'm not sure about the matrix cause there are a lot and I don't want to order the wrong one. I can't find the chips they use. 

Here are some I was looking at:

[First_64x32](https://de.aliexpress.com/item/32652594178.html?spm=a2g0o.productlist.0.0.789d1793jQfDjw&algo_pvid=f634f671-165e-4724-aec3-0f38fa4bc1d5&algo_expid=f634f671-165e-4724-aec3-0f38fa4bc1d5-0&btsid=2100bdec16045965341596981e0598&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)
[Second_64x32](https://de.aliexpress.com/item/4000030230641.html?spm=a2g0o.cart.0.0.759b3c00YlVA6r&mp=1)
[Third_64x64](https://de.aliexpress.com/item/32652209002.html?spm=a2g0o.cart.0.0.19b53c00TnUHEu&mp=1)

I hope someone knows if they will work and if I need to look at something special.

daveythacher, 2020-11-06:
For this project avoid panels based on Memory Mapped LED drivers, PWM LED drivers, FM612x drivers, etc. Pretty much anything outside your standard constant current shift registers is not likely supported or supported well by this code base.

For 64x32 I would consider Sparkfun or Adafruit. They have most of their code around valid panels. I have never heard of them stocking unsupported panels. Sadly most panel sellers do not list the LED drivers they use. It appears they order them in bulk then sell them off. The listing mostly stays the same but they sometimes change the configurations. 

The modules fit templates for LED wall configurations. The LED driver is not really a huge issue for that. Sometimes it can depending on wall module assembly traits. If you use the LED receiver/sender cards which are based on FPGAs then you can use just about any LED driver. 

On AliExpress many of them provide excellent service in terms of answering questions.

Note you will want a decent power supply. It can save you a lot of headaches. This about all I can say with out knowing about your intended setup.

jaygarcia, 2021-01-07:
@ISaneI have you made your purchases yet?

closed 2022-03-05 by ISaneI