hzeller/rpi-rgb-led-matrix #1192: Glitch cause by running desktop?

louiselessel, 2020-11-08:
Hi, 

**Does this look like a glitch caused by running the Desktop?
It happens once every 10-20 seconds or so (once in a while, not at regular intervals).**

https://www.dropbox.com/s/2nvlqxoakbse0qb/glitch.MOV?dl=0

I am building a series of python shader examples that will run on the pi (using pi3d),
and was hoping to be able to run them on the matrix as well. This is my first time using the matrixes.

Since I am running pi3d it is necessary to run a small shader window (32x32 pixel) like this (in front of the terminal)https://www.dropbox.com/s/b6ndm40gg9cybmn/running%20from%20terminal.JPG?dl=0

**I am trying to see if I can somehow run it from the terminal without initializing the desktop on bootup. I think that could be the issue, based on reading your troubleshooting section. But no luck so far... 
So before I go far down that route, I just wanted to know if that might even be the cause?**

I have messed around with all of the settings, including  `options.pwm_lsb_nanoseconds = 200, GPIO slowdown..`
and followed all of the instructions on reserving a core, disabling sound, etc. 
and I just have this one glitch left to deal with.
(Have tested several matrixes, it happens regardless).

**Setup here:** 
https://www.dropbox.com/s/x9cjhruemwpjetr/setup.jpg?dl=0
Cable length has no effect on this glitch.
Hzeller board on Raspberry pi 4.
Power supply gives (S300-5) 5.15V, 60A

Also tested Adafruit Bonnet, and Adafruit HAT.
Get a similar, but worse, more constant, glitch (https://www.dropbox.com/s/dxrmor24oliattj/Random%20lines.MOV?dl=0)

I pasted the systemctl and the top command, but I am not sure what to look for, 
so, does anything below look weird?

**MY CODE**

```
#!/usr/bin/env python

import time
import demo
import pi3d
import datetime
from rgbmatrix import RGBMatrix, RGBMatrixOptions
from PIL import Image
from PIL import ImageDraw

#-------------------------------------------------
# Configuration for the shader

(W, H) = (32, 32) # Windowed
# For scale, make sure the numbers are divisible to the resolution with no remainders (use even numbers between 0 and 1). 1.0 is full non-scaled resolution.
SCALE = 1.0 # downscale the shadertoy shader resolution

timeScalar = 0.6 # for scaling the speed of time
fps = 60 # framerate

BACKGROUND_COLOR = (0.0, 0.0, 0.0, 0.0)

display = pi3d.Display.create(window_title='shader',
                              w=W, h=H, frames_per_second=fps,
                              background=BACKGROUND_COLOR,
                              #display_config=pi3d.DISPLAY_CONFIG_HIDE_CURSOR | pi3d.DISPLAY_CONFIG_MAXIMIZED,
                              use_glx=True
                              )
#print(display.opengl.gl_id) # the type of glsl your pi is running

if W is None or H is None:
 (W, H) = (display.width, display.height)
 print('setting display size to ' + str(W) + ' ' + str(H))

## shadertoy shader stuff ##
[... 181 lines cut ...]
top - 16:55:51 up 57 min,  2 users,  load average: 0.83, 0.41, 0.38
Tasks: 174 total,   1 running, 173 sleeping,   0 stopped,   0 zombie
%Cpu(s):  9.3 us,  2.7 sy,  0.0 ni, 88.0 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   3776.5 total,   2449.9 free,    625.4 used,    701.3 buff/cache
MiB Swap:    100.0 total,    100.0 free,      0.0 used.   2781.6 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                      
 2965 daemon    20   0  152884  59052  34768 S  40.1   1.5   0:54.87 python3                                                                                      
  501 root      20   0  150880  71788  51372 S   6.6   1.9   1:59.82 Xorg                                                                                         
  903 pi        20   0  167468  49088  38432 S   2.6   1.3   0:08.35 lxterminal                                                                                   
 1363 pi        20   0   86036  38636  14216 S   2.6   1.0   1:10.55 thonny                                                                                       
  192 root      -2   0       0      0      0 S   1.3   0.0   0:27.00 v3d_render                                                                                   
  191 root      -2   0       0      0      0 S   1.0   0.0   0:19.87 v3d_bin                                                                                      
 2520 pi        20   0  590812 231116  81400 S   0.7   6.0   1:31.76 chromium-browse                                                                              
 3002 pi        20   0   10416   3152   2620 R   0.7   0.1   0:00.09 top                                                                                          
  346 root      20   0   13148   5880   5200 S   0.3   0.2   0:00.67 systemd-logind                                                                               
  960 pi        20   0  543380 141052  85204 S   0.3   3.6   1:25.78 chromium-browse                                                                              
    1 root      20   0   34712   8172   6492 S   0.0   0.2   0:03.91 systemd                                                                                      
    2 root      20   0       0      0      0 S   0.0   0.0   0:00.01 kthreadd                                                                                     
    3 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 rcu_gp                                                                                       
    4 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 rcu_par_gp                                                                                   
    8 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 mm_percpu_wq                                                                                 
    9 root      20   0       0      0      0 S   0.0   0.0   0:00.47 ksoftirqd/0                                                                                  
   10 root      20   0       0      0      0 I   0.0   0.0   0:02.91 rcu_sched 
`

jaygarcia, 2020-11-24:
Have you considered limiting the Linux OS to 3 Cores and dedicating one to this library?

xuniuer, 2021-01-07:
> Have you considered limiting the Linux OS to 3 Cores and dedicating one to this library?

How to do? Could you provide more details?
I find SSH communication from Host to RPi will cause sudden faint flicking.

jaygarcia, 2021-01-07:
Hi @xuniuer ,

edit `/boot/cmdline.txt` , add `isolcpus=3`, reboot.

jaygarcia, 2021-01-07:
Another option may be to overclock the pi a little? Just a thought.

xuniuer, 2021-01-08:
> Hi @xuniuer ,
> 
> edit `/boot/cmdline.txt` , add `isolcpus=3`, reboot.

I consult the settings of `isolcpu=3` as refer: https://yosh.ke.mu/raspberry_pi_isolating_cores_in_linux_kernel
As understood, only a process is moved onto the core, user-process time would not be stolen - RTOS.

However, in my case, some data exchange (such as `redis subscribe`) happens in order that RGBMatrix shows what I want - some gevent tasklets running (using python bindings with gevent support). Everything is in one PROCESS. Now I find that limit refresh rate to 300 or less is better. Without limit of refresh rate, refresh rate is up to 340+ with 8 chains and 2 parallels based on 3B. 

Maybe try to use multi-processing and psutil libraries to balance them.