hzeller/rpi-rgb-led-matrix #1194: Create Matrix Configuration for Dummies

daveythacher, 2020-11-10:
There are a few users on this project that seems to be really struggling with how this project works. Lets go over a few things done wrong:

1. Fails to understand the LED drivers are really just SPI or 4 wire serial. These likely are built on CMOS and will have a max frequency of around 25-50MHz.
2. Fails to understand BCM or PWM in a meaningful way. Binary is a number representation system where place denote magnitude just like decimal. This means for every place its value doubles.
3. Fails to understand that these are serial shift registers which need time to move the entire column over before a single place can begin. (Can occur concurrently, which it does. This is implementation detail.)
4. Fails to understand that OE time is this time unit. Yet sets it to 100ns because it gives amazing performance. (May violate OE pin of LED driver timing.) However does not ensure that each place is given the correct weight of time. Which is required for PWM/BCM period to be accurate.
5. Fails to understand that they are using it in a mode which allows it to be incorrect within a range not bound by binary places.
6. Fails to understand the issue of driving the signals too fast in terms of frequency domain of drivers and wiring.
7. Fails to understand the issue of not using correct power for heavy peak load and what this can do for electronics on the same rail.
8. Does not understand how the BCM windows are accumulated. This can be a limiting factor however it was actually designed this way to help with certain things. (Still a little crazy.)
9. Fails to understand performance and display size limits.
10. Fails to understand that dithering does not matter depending on point 5. Dithering reduces quality assuming not already reduced. However can remove wasted work, which will case speed boost to reported FPS.
11. Fails to understand that dithering is a interlaced like trick for reducing quality during select multiplex scans. Thus reducing the average time per frame and increasing FPS.
12. Fails to understand what reported FPS really is. Fails to understand what bit banging can do in this context. (Eg not like running "time ./prog" in shell.)
13. Fails to understand the implications of software for IO processing in terms of saturation and rate; user space processes, bit banging, cache stability, etc. This can effect setup size, stability and quality.
14. Many many more off topic things.

BCM is done to save computation time. It is really not useful outside software. Then again you really should be using a different class of processor given the IO nature of this, which is very taboo for software. Well Turing/Assembly style software. Software is ROM bits which configure hardware to performance operations. You can do work directly from the ROM bits.

Attitude even from developer was that BCM did not have time window. Which is actually true depending on usage, however this requires special consideration. PWM hardware is used to save memory and avoid OS instability. (It is really a timer, but the user has to configure it with command line. No useful documentation exists to my knowledge.) A decent number of tricks are used by this project to try to make a processor like the Pi work in an area it is poorly suited for. However this mostly has to do with the class of LED driver and cost optimization. 

For smaller configurations that are not super demanding, this project will be workable for personal, education, etc. purposes. Even some commercial if great concern is considered. (If a hacker gets you, well there goes your display.)

So there is two options for configuration. Critical pwm bits and non-critical pwm bits. Critical pwm bits is already outlined in another issue #1120 . It requires a little computation. Non-critical is really easy but you have to understand it. This one will pretend to give you more PWM steps than you actually get. It will also report really high frame rates. 

The high frame rate condition is used to slip past binary places, meaning you get less steps per color in reality than advertised. The higher bits are more correct than the lower bits. This give a boost in performance but the quality is less in terms of detail on the finer color steps. Some may not be able see a difference beyond a point and some applications may not need such detail. See [RGB Palettes](https://en.wikipedia.org/wiki/List_of_monochrome_and_RGB_color_formats#Regular_RGB_palettes).

For really large setups per connector you have to use the critical pwm bits. Again this takes some computation to configure correctly. (Why I am not sure.) I do not know how the Pi performs in this situation. However users have reported limits which are not consistent with serial limits or library implementation.

This project suffers greatly from terrible and non existent documentation. I refuse to commit to this project beyond issue tracker feedback.

jaygarcia, 2021-01-13:
@daveythacher this seems like an awesome addition to the Wiki =).

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828