hzeller/rpi-rgb-led-matrix #1196: 2020 P10RGB-3535-4S driving problem

hakao32, 2020-11-11:
Hello,
I have a problem with 2020 P10RGB-3535-4S LED Matrix. 2019 P10 was working with 
`--led-rows=16 --led-cols=32 --led-chain=6 --led-parallel=2 --led-multiplexing=2`
Yet 2020 one not working. I get the best result by:
`--led-rows=16 --led-cols=32 --led-chain=2 --led-row-addr-type=0  --led-no-hardware-pulse --led-multiplexing=8 --led-slowdown-gpio=3` and trying to write "onair":
[image]

Can u pls help?

daveythacher, 2020-11-12:
This looks like one of the outdoor panels which use a long shift register to avoid lowering average current reductions in multiplexing. This is done to make them brighter. There is an summary for how to do this here: #1190 .

There is technically two ways to do this. One is with a pixel-mapper where you edit MapVisibleToMatrix and GetSizeMapping. The other is with multiplex-mapper where you edit MapSinglePanel and GetSizeMapping. 

I recommended pixel-mapper in #1190, which may not have been correct. However the code base may not really care in this case. Multiplex-mapper really is a pixel-mapper with a special function that is not used to my knowledge. Multiplex-mapper is supposed to be used for fancy row scan approaches of multiplexing.

The problem is these are not that. These are really long serial chains wrapped around internally. To me that suggests pixel-mapper. On the command line you will need to tell the library how long the actual chain is. This will be different than the panel size. GetSizeMapping will convert this to the panel size for you, so that your logic can work with the panel size versus pixel representation.

Technically you could update that special function, EditColsRows. The input is panel size which gets converted to chain size. So now on the command line you specify the actual panel size. Which gets converted to chain size via EditColsRows. Then converted back to panel size with GetSizeMapping. Now the internal library works with chain size while input works with panel size via MapVisibleToMatrix.

Pretty crazy right? Wait till you see how application specific those classes will be. There is already a decent sized list of classes for multiplex-mappers, pixel-mappers, RowAddressSetters, etc. Most are basically the same, just with minor changes.

@craiglee2 @merliongold You may have been on the right path. See above.

daveythacher, 2020-11-12:
Pixel-mappers I think have configuration parameters. While Multiplex-mappers do not. Therefore the ability to standardize classes is nothing. Hey that sounds like an enhancement...

craiglee22, 2020-11-12:
If you're using static 1:1 duty, then you can use lookup table which is remap the pixels lower rows to upper rows vice versa. x = LUT[x][y]; y = 0, 1, 2, 3 based on y-axis in my case. Please make sure using minimal-example.cc in example-api-use folder width x height Canvas->SetPixel(0..width-1, 0..height-1) to debug right places.

hakao32, 2020-11-12:
Do you want me to do [this](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1190) then this:

> If you're using static 1:1 duty, then you can use lookup table which is remap the pixels lower rows to upper rows vice versa. x = LUT[x][y]; y = 0, 1, 2, 3 based on y-axis in my case. Please make [...]

?
I dont have such info about my RGB matrix like pixel mapping yet I will try.

daveythacher, 2020-11-12:
The implementation in #242 is based on transformers, which is no longer supported.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/84e1465e9ea5ed000011d05369c5287eaa361ad7/include/led-matrix.h#L244-L246

hakao32, 2020-11-12:
I made it work. I clone all repo and I made it running with --led-multiplexing=4. However I have another problem witn latest version of lib.  Latest version does not work with [modifcation](https://github.com/hzeller/rpi-rgb-led-matrix/issues/527). Yet it is not relevant to this issue.

daveythacher, 2020-11-12:
> This looks like one of the outdoor panels which use a long shift register to avoid lowering average current reductions in multiplexing. This is done to make them brighter. There is an summary for how to do this here: #1190 .
> 
> There is technically two ways to do this. One is with a pixel-mapper where you edit MapVisibleToMatrix and GetSizeMapping. The other is with multiplex-mapper where you edit MapSinglePanel and GetSizeMapping.
> 
> I recommended pixel-mapper in #1190, which may not have been correct. However the code base may not really care in this case. Multiplex-mapper really is a pixel-mapper with a special function that is not used to my knowledge. Multiplex-mapper is supposed to be used for fancy row scan approaches of multiplexing.
> 
> The problem is these are not that. These are really long serial chains wrapped around internally. To me that suggests pixel-mapper. On the command line you will need to tell the library how long the actual chain is. This will be different than the panel size. GetSizeMapping will convert this to the panel size for you, so that your logic can work with the panel size versus pixel representation.
> 
> Technically you could update that special function, EditColsRows. The input is panel size which gets converted to chain size. So now on the command line you specify the actual panel size. Which gets converted to chain size via EditColsRows. Then converted back to panel size with GetSizeMapping. Now the internal library works with chain size while input works with panel size via MapVisibleToMatrix.
> 
> Pretty crazy right? Wait till you see how application specific those classes will be. There is already a decent sized list of classes for multiplex-mappers, pixel-mappers, RowAddressSetters, etc. Most are basically the same, just with minor changes.
> 
> @craiglee22 @merliongold You may have been on the right path. See above.

Actually you could just pass the number of rows divided by multiplex scan into stretch_factor for MultiplexMapperBase. If you do that then you would not need to change EditColsRows or GetSizeMapping. However you would need to create a new Multiplex-mapper and register it with the code base. Kind of crazy.

daveythacher, 2020-11-12:
> I made it work. I clone all repo and I made it running with --led-multiplexing=4. However I have another problem witn latest version of lib. Latest version does not work with [modifcation](https://github.com/hzeller/rpi-rgb-led-matrix/issues/527). Yet it is not relevant to this issue.

I do not understand.

daveythacher, 2020-11-12:
@craiglee22 Do you want to contribute your code since is looks like there is no single multiplex mappers. Converting the PixelMapper logic to MultiplexMapper logic looks straight forward.

There is no way that is sane on the command line. I had no clue what spiral and z mappings were. That was a mystery to me. I kind of wish I was still in that state to be honest. No wonder people just try them in rapid fire. I even read the README.

craiglee22, 2020-11-12:
Ok how to upload each files?

daveythacher, 2020-11-13:
You need to create git branch. Commit your changes to the branch and push it all to the remote server. Then create a pull request via github. It will get review and merged in. Include #1190 in commit message.

git stash
git checkout -b 1190
git push origin 1190
git pop
git commit -a -m "#1190 adding Multiplex Mapper for 16x16 1:1 scan outdoor panel"
git push

hakao32, 2020-11-17:
> I do not understand.

I was using old version of the repo. I reclone it and use it for driving LED. There is no need to make changes for 2020 P10RGB-3535-4S. I used basically  --led-multiplexing=4 to make it work properly.

daveythacher, 2020-11-17:
So yours is working? Are you using this: https://github.com/istaiko/rpi-rgb-led-matrix or this https://github.com/hzeller/rpi-rgb-led-matrix

daveythacher, 2020-11-17:
I am not sure why you need the older version. Since I think it is the same. Yours is likely a standard 1:4 outdoor. I assumed you tried all of them. Post new issue for unrelated issue.

craiglee22, 2020-11-17:
Git push origin 1190
fatal: remote error
  You can’t push to git://github.com/hzeller/rpi-rgb-led-matrix.git
  Use https://github.com/hzeller/rpi-rgb-led-matrix.git

daveythacher, 2020-11-17:
git remote set-url origin https://github.com/hzeller/rpi-rgb-led-matrix.git

daveythacher, 2020-11-17:
Try git push -u origin 1190 first

craiglee22, 2020-11-17:
Please make sure it is okay...

closed 2021-09-23 by hakao32