hzeller/rpi-rgb-led-matrix #1199: GPU use

hakao32, 2020-11-17:
Is it possible to force the LED driving scripts use GPU? Or is it already handled at drawing canvas etc?

daveythacher, 2020-11-17:
You can draw pictures with the GPU. However the CPU will have to call setPixel for each pixel within the frame. This really complicates things too. Normally you would let hardware do this. 

If you had hardware for updating display you could either have hardware read the external video out signal directly. Or you use internal frame copy from memory to hardware. Which could allow DMA. The hardware would have some memory which holds the frames just like a GPU or monitor does. (Think VGA)

Receiver and sender cards allow this. So you just write a GPU application which gets rendered by the GPU. This is sent via HDMI to the sender card which processes it. This is sent to the receiver which drives the LED panels. This makes sense for video walls, but is not very compact and energy efficient of other applications. Can get costly too depending on certain things.

setPixel is a fairly expensive operation. There is a couple schools of thought on this. It really does not matter. There are a couple cases where it looks it may. However to avoid slow downs you can use multiplex threads to update. You can use multiplex canvases as well. Now this may destabilize the Pi. 

The Raspberry Pi is a low cost general purpose micro-processor using shared L2 cache. This means despite so noteworthy attempts to optimize this code base for the Pi, issues can still arise. It would depend what you are trying to do and how large the setup is.

In summary yes you can use the GPU to generate frames. However there is a slight cost for doing so. It will work better for static frames. For videos you may want to use multiple canvases and/or multiple threads.

hakao32, 2020-11-17:
I was thinking to put heavy load to GPU to prevent flickers when there is lots of back ground tasks that uses CPU. 
I dont even know that is it possible to run specific function at GPU while others in CPU or not.
There is some known ways to use GPU like DRI_PRIME and CUDA yet I guess they are not suitable for Raspberry Pi.
For python binding there is numba to use GPU, is there any one who try this before?

daveythacher, 2020-11-18:
I am going to recommend against this premise. Using the Pi with Python and GPU compute along with this library. This will not work as well as you think. There is a few issues worth mentioning here: Python, Pi, Locality, OS, etc.

Python is a very useful for scripting. It has the flexibility of a programming language with the flexibility of a scripting language. However it is not as fast. It starts up slow, but for certain things it is really fast. Most of the file operation are wrapped around C logic. So Python does perform decently with files. This is useful on Linux since a lot of the configuration to the kernel is well files. Linux uses the file system for a decent amount of configuration. Sockets are files. Syscalls are not which represents another major interface. File system calls are usually syscalls.

Python does not have proper support for multithreading and uses C style forking for multi process as I recall. It uses operating system notions for inter process communication if I recall correctly. One of the advantages to monolithic kernels. They tend to be general purpose in nature. Python tends to be the same.

Python has a major appeal in configuration logic. The logic is flexible and portable. However is not very runtime focused. However can be very fast. Python can wrap up heavy calls but Pythons will create latency in the program. Python is a serious security risk also.

The Pi is a low cost general purpose computer. This allows many options. It supports operating systems and other code bases which allow more complex logic to be expressed with minimal effort. However the Pi is designed like a micro-processor and tends to favor batch computation. Batch computation is a does not model itself after events but rather raw work throughput.

This means events or processing between workloads is not ideal. This creates issues. Normally what you can do is speed it up really fast to make the problems go away. This works but since clock speeds and architectural gains slowed this has not been effective. This also increases power consumption.

The Pi memory system runs against single memory for CPU and GPU. The cache is shared between cores in L2. Meaning if a stall condition occurs due memory access blocking then time boundaries will slip. There is little you can do about this.

However if willing to use external hardware you can create a notion of locality. The hardware events are tracked and managed directly in parallel. Not much options exist here for the Pi. However if you included external hardware you can allow slippage of frame draw while multiplexing is not effected. Dual buffered memory can be used to avoid tearing. Dual port memory can be used to avoid contention or bus stalls.

Operating systems usually use ticks to determine which process should run. These ticks are fairly slow. This is done so that each process has enough time to wake up and does something. A decent amount of overhead is involved in context switching in very large software notions. These notions are used to create general purpose.

The kernel does support interrupts. However this can still involve a decent amount of overhead. Effectively there is latency in processing. Now multicore systems can improve this however this creates inter process and memory stability issues. To reduce the latency really fast processing can exist. However the issue comes back to memory pressure. You can relieve memory pressure but it creates problems for the programming model.

Therefore general purpose systems can be used however have many pitfalls. They excel at certain things and are really poor at everything else. With proper system design you can avoid all of these pitfalls. The cost argument has slowly been disappearing. The time argument still exists mostly due to Java and Python.

This creates issues for mobile markets. Mobile is an interesting one since no one really knew how to make the programming model work and get the power consumption under control. Cost was a problem for a long time. Making fast and low power processors was once a manufacturing issue. This has disappeared.

Lot of people still look at the processor cores for system and architectural performance. However just like Python these processor cores are configuration components. They to some extend do not matter. This is the detail between AVR and PIC. This detail also exist heavily in the ARM world. It is a conceptually outdated view.

This allowed for the rise of IP processor cores. It was no longer worth the time to build your own processor core, compiler, operating system, etc. Now many security bugs came from this. They did not understand that custom logic would still remain. Instead they tried to delete all custom logic.

There are many processors out there which use small "real time" processors on board. They are on the same die, but are completely screwed in terms of real time due to memory subsystem. Inter-process communication remains an issue there. One could simply build a FIFO. If one is familiar with operating systems and multi-processors this should be clear.

It would appear you plan to do compute not graphics with GPU. This can cause memory pressure in cache and main RAM. Depending on complexity of computation and size of display this can cause serious issues for this library. You will be doing a number of memory copies to move data in and out of GPU. There will be other system processes. There could be multiple threads doing setPixel/update processe [... cut ...]  may not have anything to do or starve a critical task. The operating system may also attempt to protect it.

Software is usually not good at IO. Hardware however is. The irony of the memory wall is that it is all IO. Software was supposed to shift gears, but appears to never have. There is a lot of ways for software to exist. These do not cause issues with silicon economics as much as you would think. The problem ironically has been on the software side. Maybe a little of the circuit area also.

jaygarcia, 2020-11-25:
@daveythacher Dude! thank you for taking the time to contribute to the global knowledge pool here! It's clear that you're passionate about technology and have a solid understanding of lower level electronics and how it ties to software written! To be honest, the fact that you're taking the time to explain stuff like this in such great detail is dope! Thank you dude!!

closed 2025-11-27 by marcmerlin