hzeller/rpi-rgb-led-matrix #1201: What RPi for two 64x32 panels

Boerny41, 2020-11-18:
First of all this is not an issue but more like an issue prevention question.
I'd like to know what RPi is best used for these panels. I know the guide says all work but it does not seem to factor in the newer ones.
I read here that the RPi 4 is to fast and that it leads to problems. However recent RPi 3s are nearly the same speed - 1.3Ghz instead of 1.5Ghz iirc.
I guess the guide is talking about the very first RPi 3s and I'd also guess that they are alot slower. However I'm obviously not sure and now don't know what to buy to make it work...

daveythacher, 2020-11-19:
I was also under the impression the Pi 3 was lower in clocks. The Pi's have avoid high clocks speeds for a long time. I know that they jumped into quad cores quickly. They also underclock quite a bit. The Pi part line in a fricking mess. 

The Pi foundation is off its rocker for good this time. The should have started with the Pi 4. The desktop performance of the Pi sucks, and honestly that was its job. I am tired of seeing people screw up microprocessors and confuse them with microcontrollers. Bit banging was a mess in Arduino and there it could make more sense. Also hopefully steer you toward better ways...Arduino never really tackled that. The Pi certainly made it worse. Then Python...oh how you fallen.

The Pi changed clocks, IO configurations, memory footprints, etc within versions. Most of the times the processor is an architecture change. This gives a boost in performance most of the time. While rebuffing power consumption and thermal issues. Apparently the ARM cores can actually thermal throttle. They do not boost but will thermal throttle.

The Pi 4 is likely a huge change in terms of architectural performance and clock. Together these can create thermal problems, which can come up in weird ways. The issue however to also note is that this code base is technically doing something that you never really want to do in this type of processor.

Assembly style software is really poor at IO manipulation. ROMs are better, but gates are ideal. This project does a lot of bit mangling in software and does bit banging on IO. It uses a few optimizations but still struggles due to a number of issues. The shared cache, unbound IO operations, unbound core clock, dead loop block, etc.

Now this assumes the core became really strong and basically the ability toggle the pins rises. Another potential issue is the shared RAM of the GPU becomes a potential issue if it got an upgrade too. The RAM got large which could have made is slower to. Rising the clocks and actors on it will force more pressure. This could in turn lead to performance problems and thermal problems.

Few people have issue with the Pi 3. Overall I would not recommend the Pi. It was clear from day one it was interested in targeting a cheap desktop general purpose computer. It was created for playing around with programming. You really should be using a microcontroller or FPGA for this.

Technically you can use a microprocessor running a loose notion of "real time" or non real time operating system. This would be a function of the underlying hardware. If the hardware does not exist you can use a FPGA to extend it. However the Pi is IO constrained on interfacing.

For small applications you may be able to direct drive. However for larger ones you really are not sane for using a Pi. However if you really understand this library you can possibly pull it off. There is a number of games you can play.

A word of caution. For small display setups with modest configurations will fare much better. If you are going to push this to the limit KNOW WHAT YOU ARE DOING! The configuration of this library is very backwards. There is a school of thought behind it but it is not required per say. If you understand the proper configuration procedure that will save you a lot of headaches.

Misconfiguration can stress the processor and reduce the quality. It is worth mentioning that this project will drive a full core at full usage. You will have the thermal for a full core and there is nothing you can do about it. The operating system and microprocessor kind of make this a catch 22.

The Pi 4 handles desktop a lot better now. However you probably want to disable this. Again the RAM is shared. CPU probably wins in contention, but there could be DDR RAM in there now. Which means there could be latency. This does not matter for slow processes or non real time processes. However for multiplexing this is a very much required.

You know there could have been an issue with the main RAM with DMA to IO. It could also be the DMA limitation or IO bridge. Yet that did not stop anyone from making outlandish statements about DMA being "too slow". As I recall a lot of memory transactions take place on wide buses to overcome latency. DMA would likely work at this width maybe double depending on clocking scheme.

I view certain aspects of this project as dangerous. Overall I think it has some merit, mostly for smaller projects and people who understand it. However it makes statements without full context that quite frankly are designed to look stupid and potentially spread misinformation. It also has a few which are a little on the wild side. Happy to mock certain things and attack aimlessly. (Aka fools who look like trolls or experts, they will probably enough me saying that.)

Boerny41, 2020-11-19:
Sorry that i write just this short thing after you gave this much info but what Pi is good to use now? I know for certain that I want to go the RPi route instead of using a microconroller but the question is still the same: the Pi 3 or 3+ or model B super duper plus? maybe the Pi 4 or rev. B 4? I have no clue...

hzeller, 2020-11-19:
Don't mind David, he likes to troll around and confuse people with walls of text to look knowledgeable.

For your project, any Pi 3 or Pi 4 will do. For the Pi4, you might need a higher [`--led-slowdown-gpio` value](https://github.com/hzeller/rpi-rgb-led-matrix#gpio-speed), which addresses the 'too fast' problem. Other than that things will be fine. In general, the Pi4 is better as you have more capacity for other things going on.

For discussions and general advice, I'd like to point to https://rpi-rgb-led-matrix.discourse.group/ which was set up by @marcmerlin

Boerny41, 2020-11-19:
Ahh thanks sounds good. Didn't know about the group gonna check it out :)

closed 2020-11-19 by Boerny41