hzeller/rpi-rgb-led-matrix #1203: Control the led matrix/the rPi with home assistant to display a new image when certain conditions are met

Boerny41, 2020-11-21:
Hey guys,

I wondered if it is possible to controll the rPi (which is controlling the matrix) via home assistant. Since it is possible to control the rPi/matrix wirelessly it should be possible this way to... right?
For example home assistant knows the weather of you location and is able to turn a smart bulb on when the weather changes. Now I don't want a bulb to turn on but a signal send to the Pi so it loads shows another image on the matrix e.g. rain clouds once HA notices rain.

I know that this is not really the right place to ask but the HA forum seems to be not as active and my this question got ignored there... :( 
All I was able to find out by now is, that HA is able to controll the GPIO pins of a rPi but I'm not sure if that is helpfull at all^^

hzeller, 2020-11-24:
This might probably more something for the discussion groups on https://rpi-rgb-led-matrix.discourse.group/

I think this is more a generic question and indeed would be probably more a HA question. Ideally, you'd send a network packet to your Pi to trigger it to display something.
I don't know what common things are that home assistant is doing, but I suspect something like that should be doable.

Boerny41, 2020-11-24:
I'll check the group out thanks for the link. HA forum surly is a better place to ask but as I said they either don't know or don't care :/

closed 2020-11-24 by Boerny41