hzeller/rpi-rgb-led-matrix #1204: Image-Viewer Image refreshing

git-clone-nick, 2020-11-23:
First of all, Im fairly new to the whole led-matrix thing and very thankful for the great library.

I just want to know whether it is possible to have the image viewer refresh the given image every second or so.
Essentially Im trying to show the amount of parking-spots that are left in a parking garage. I would use a normal font and display the changing numbers there but I also need pictures like arrows and a big blue P on the same matrix. So my Idea was to have single Images of numbers and symbols, which are put into one image through the magick script and to then show the whole image with the image viewer. The image is supposed to change with every free or taken parking-spot. 
[image]

any help would be very much appreciated.
Greetings from Germany.

jaygarcia, 2020-11-25:
@git-clone-nick , one easy thing to do is kill your the program and restart it. 

the more elegant approach may be to clone the image demo code and use something like inotify to reload the file into memory upon change.

git-clone-nick, 2020-11-30:
> @git-clone-nick , one easy thing to do is kill your the program and restart it.
> 
> the more elegant approach may be to clone the image demo code and use something like inotify to reload the file into memory upon change. https://man7.org/linux/man-pages/man7/inotify.7.html

Thanks for your answer @jaygarcia , sadly killing and restarting is no option for me, because it leaves a split second of just a black screen.
I will try your other suggestion though.

git-clone-nick, 2021-02-15:
I have found a solution to my problem!
It was as simple as changing to the python binding.

closed 2021-02-15 by git-clone-nick

SJoWie80, 2021-09-08:
I want to do something similar. but with an image url (spotify album art)
do you have some examples?

git-clone-nick, 2021-09-10:
> I want to do something similar. but with an image url (spotify album art)
> 
> do you have some examples?

So you mean you want to get it from a web server? That would be as simple as using the python binding and the http requests. For further questions just email me mail@nickmaier.com