hzeller/rpi-rgb-led-matrix #1207: Led panel P5-6432-3528-16S-B4 with ICN2038s chip

maksun, 2020-11-26:
Hello,

I just would like to know if someone here managed to make this panel working?
I'm using the active board from electrodragon, I'm trying to help a friend so I did some research and read the following:

"Check the controller chip that install the LED panel, if are ICN2038S or FM6126, those are not compatible Smartmatrix because these chips require three clock pulses by Latch."

I will tell my friend to try the led panel type FM6126 in case of... Any advice is welcome ;)

Thank you.

maksun, 2020-11-27:
Hello,

Looks like this chipset is similar to the FM6126A as I managed to make it working with the "--led-panel-type=FM6126A" parameter. FYI, it was an ABCD panel style.

Regards,

alsaleem00, 2021-01-05:
Hello I am able to display something on matrix. but refresh is bad.
If anyone is interested I can post more here

wrldtvlr, 2021-02-23:
I have a P2 panel with this same chip. While I can display something on the panel, the addressing isn't right I would be interested in knowing more about what you've been able to do.

alsaleem00, 2021-02-23:
most P2's have different different row driver (320x160mm has 1/40 scan). This requires handling the row driver in addition to ICN2038s. please check your matrix. 
If your matrix is 256x128mm then you only need the ICN2038s driver (scan will be 1/32 which is OK.)

wrldtvlr, 2021-02-23:
> most P2's have different row driver (320x160mm has 1/40 scan). This

Yes, physical size is 256x128, with 128x64 pixels and 1/32 scan.

alsaleem00, 2021-02-23:
here is the code 
#define BITS_ON	MTX_RGB_PINS
#define BITS_OFF	0
#define ICN1038S_REG_1		11
#define ICN1038S_REG_2		12
#define ICN1038S_DEF_REG1	(0xf000)  // register 1. (11)
#define ICN1038S_DEF_REG2 	(0xf000)  // register 2. (12)
void matrix_write(uint16_t uVal, int width, int reg)
{
	uint16_t value;

	MTX_LAT_LOW();
	MTX_CLK_LOW();
	for (int i = 0; i < width; ++i) {
		value = (uVal & (0x8000 >> (i%16))) ? BITS_ON : BITS_OFF;
		if (i > (width - reg)) {
			MTX_LAT_HIGH();
		}
		PIN_LOW(MTX_RGB_PORT, BITS_ON);
		PIN_HIGH(MTX_RGB_PORT, value);
		MTX_CLK_LOW();
		MTX_CLK_LOW();
		MTX_CLK_LOW();
		MTX_CLK_HIGH();
		MTX_CLK_HIGH();
		MTX_CLK_HIGH();
	}
	MTX_CLK_LOW();
	MTX_CLK_LOW();
	MTX_LAT_LOW();
}
void icn1038s_init()
{
	matrix_write(ICN1038S_DEF_REG1, LED_WIDTH, ICN1038S_REG_1);
	matrix_write(ICN1038S_DEF_REG2, LED_WIDTH, ICN1038S_REG_2);
}
I repeated the clock/latch signals because my MCU is fast.

jeroenverspeek, 2024-10-19:
@alsaleem00 looks like the right solution for the ICN2038S. I am trying to get a 128x128 panel to work with this chip (https://github.com/hzeller/rpi-rgb-led-matrix/issues/1713). 

Your code does not look like it is written for hzellers library. If you could give more (code) context that would be helpful.

alsaleem00, 2024-10-19:
This code was written for STM32F722. 
The PIN_/MTX_CLK_/MTX_LAT_/MTX_CLK_to control the pin directly.

jeroenverspeek, 2024-10-19:
Hi Alsaleem00,
Thanks for your reply.

closed 2025-11-27 by marcmerlin