hzeller/rpi-rgb-led-matrix #1210: Raspberry Pi Zero W Raspian Stretch System Stall with 16x32 Matrix Running rpi-rgb-led-matrix

ebarlas, 2020-12-04:
Hi folks, I'm experiencing a bizarre issue.

I have simple C++ LED matrix project built atop the rpi-rgb-led-matrix library: https://github.com/ebarlas/wave-tracker-led-matrix

After running for several hours, my entire Raspberry Pi Zero W system with Raspian Stretch locks up. Everything seizes up and mysteriously resumes about an hour later. Power isn't lost. In fact, when the system becomes responsive again, it's as if nothing happened. The processes are still running and the rpi-rgb-led-matrix goes right back to scrolling lovely images and text.

The problem appears to be specific to the LED matrix with rpi-rgb-led-matrix. When I disconnect those components, the system is as stable as can be.

Has anyone encountered this?

Could it be a heat issue?

Interference?

To demonstrate the problem, the following background script is running:
```while true; do date; sleep 10; done >> test.log```

Here's a snippet from the log file during this mysterious stall this morning:
```
Fri Dec  4 01:24:22 PST 2020
Fri Dec  4 01:24:32 PST 2020
Fri Dec  4 01:24:42 PST 2020
Fri Dec  4 01:24:52 PST 2020
Fri Dec  4 01:25:02 PST 2020
Fri Dec  4 01:25:12 PST 2020
Fri Dec  4 02:36:56 PST 2020
Fri Dec  4 02:37:06 PST 2020
Fri Dec  4 02:37:16 PST 2020
Fri Dec  4 02:37:26 PST 2020
```

hzeller, 2020-12-04:
It might be CPU contention; check how much CPU utilization you have on the matrix-updating thread (use `top` and press uppercase `H` to see threads).

If it is 100% or close to it, then the single-core Pi Zero can lock up as then the realtime thread uses too much CPU; anything else then might bring the system into trouble. Single core and a realtime thread can cause that as there is no other available core when some more work is needed.
A single 16x32 can do that, in particular if it has 1:4 multiplexing as it results in a particular high refresh rate.

I suggest to increase [`--led-pwm-lsb-nanoseconds`](https://github.com/hzeller/rpi-rgb-led-matrix#misc-options) until you reach a CPU utilization of that thread that is more in the 80%-90% range.

Also: From user reports, as suggested operating system distribution, the more lean DietPi does seem to behave better overall.

ebarlas, 2020-12-04:
I had the same thought earlier but I didn't notice anything noteworthy.

Thread CPU is currently in the 40-50% range.

I wonder what the something else could be? I removed the recommended packages. Maybe I'll try DietPi to see if the problem subsides.

ebarlas, 2020-12-05:
I just finished switching from Raspbian Stretch Lite to DietPi. Fingers crossed!

ebarlas, 2020-12-05:
The issue has not recurred. Thank you, again, @hzeller! I'm closing the support ticket.

closed 2020-12-05 by ebarlas