hzeller/rpi-rgb-led-matrix #1215: Error learning basic examples

jax200, 2020-12-17:
I am new to python so any help is appreciated...

pi@raspberrypi:~/rpi-rgb-led-matrix/bindings/python/samples $ python runtext.py
Traceback (most recent call last):
  File "runtext.py", line 3, in <module>
    from samplebase import SampleBase
  File "/home/pi/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/home/pi/rpi-rgb-led-matrix/bindings/python/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
ImportError: No module named core

pi@raspberrypi:~/rpi-rgb-led-matrix/bindings/python/samples $ python samplebase.py
Traceback (most recent call last):
  File "samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
ImportError: No module named rgbmatrix

hzeller, 2020-12-17:
Did you install the Python bindings first ? https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python

jax200, 2020-12-17:
I tried but getting stuck with this...
pi@raspberrypi:/ $ make build-python
make: *** No rule to make target 'build-python'.  Stop.

hzeller, 2020-12-17:
Again, make sure you're in the right directory.

jax200, 2020-12-18:
I am in the root directory as per instructions.

hzeller, 2020-12-18:
In the root directory of the library. So whereever you have checked out rpi-rgb-led-matrix.

jax200, 2020-12-18:
So in this location?  
pi@raspberrypi:~/rpi-rgb-led-matrix/lib 
or
pi@raspberrypi:/lib 

Thx for your help, I am still learning..

hzeller, 2020-12-18:
It would be here

```
pi@raspberrypi:~/rpi-rgb-led-matrix
```

jax200, 2020-12-18:
Yes, all good  :  )

thx so much for your help!

jax200, 2020-12-18:
In the image-scroller.py example, I put in -r 16, but it is cutting it in half.

hzeller, 2020-12-18:
I'd first suggest figuring out all the settings using the ready programs, such as the image viewer or text scroller, where you can play around with the [command line flags](https://github.com/hzeller/rpi-rgb-led-matrix#changing-parameters-via-command-line-flags) that have all options accessible.

Then moving on to the python examples where you might not have all command line options, but can set them easily in the option struct when you construct a matrix.

hzeller, 2020-12-19:
For general questions that are not bugs, probably https://rpi-rgb-led-matrix.discourse.group/ is a good forum ?

jax200, 2020-12-19:
Thanks for this.   I'll move over there with my questions.

closed 2020-12-19 by jax200