hzeller/rpi-rgb-led-matrix #1217: web controlled rgb-led-matrix

muharramapri, 2020-12-22:
currently i am working on my final project as a student with the Raspberry Pi, and since i am a beginner i am having some difficulties. The project is basically: 80x40 Led RBG Matrix Panel connected via GPIO to RasPi I downloaded this library (https: //github.com/hzeller/rpi-rgb-led-matrix) to control it and - after a lot of trouble - it worked with pretty good. I can also place images and text and the results are really nice to look at. whenever the instructions in the terminal (sudo ./led-matrix -d -t ...) are not completely user friendly. So the idea - what my lecturer recommended was to create a web server with a graphical interface, which could display for example a list of all the images available for display, so that it could be accessed via mobile phones too. The next step is to allow users to directly upload their own image / text. Can you help me

hzeller, 2020-12-22:
This is probably better discussed on https://rpi-rgb-led-matrix.discourse.group/  ; closing this issue here.

But in general, you need to find out how to implement the webserver yourself, this is beyond the scope of this library. Should be fairly straighforward.

closed 2020-12-22 by hzeller

muharramapri, 2020-12-22:
I apologize in advance and now I have joined the rpi-rgb-led-matrix
discussion group..thanks