hzeller/rpi-rgb-led-matrix #1219: Interest in providing a kernel mod for this library?

MarcKarasek, 2020-12-27:
A few years back I took this code and created a kernel module + application (demo). 

I also created a web server that took the place of the app and could receive commands from a remote host.  A demo app client running on another PC or an android app.  

I am looking to revive this project.  Is there any interest in having a fork of the current tree with this support?  

My initial plan was to be able to take a picture on a phone and send this to the display, but it could also be used to update a display over a local ethernet network as well.  

Marc

MarcKarasek, 2020-12-27:
If anyone is interested in my code I cloned it is at: 

https://github.com/MarcKarasek/rpi_rgb_ledmatrix-kerneldriver

You will also find the android app there as well. 
https://github.com/MarcKarasek/Android_Led_Server

xuniuer, 2021-04-04:
I guess ledmatrix in kernel mode could display something earlier than it in application mode. In application level, we just see something after `rc.local` is executed.  At least 10 secs passed (headless). More pictures or data expected here or in your branch.

MarcKarasek, 2021-04-05:
I have taken a look at this and it does not seem like a lot of work to move what I originally did over to a new branch.  I may have some free cyclles in the next few weeks to look at this further..