hzeller/rpi-rgb-led-matrix #1220: Intermittent initialization error I think

BigPeteNZ, 2020-12-27:
Hi,

I'm not sure this is the right place to randomly post this, but this site is coming up regularly in my google searches, and reading through the posts, the people here seem well qualified to help me out.

Long story short, I ended up in charge of fixing the school LED billboard. It consists of 2 arrays. Each array is 3 panels by 3 panels. Each panel has 128x64 RGB LEDS and 320x640mm in size.

A new array costs $24k, so the school is looking at trying to repair it first.

I've tried finding a new panel but can't find one the same to replace it with so I'm looking into fixing it.

The LED drivers are MBI5051GP ICs
https://datasheet.lcsc.com/szlcsc/1810311720_MBI-MBI5051GP-B_C261126.pdf

There's also a few HC245 Octal bus transceivers in the circuit
https://www.ti.com/lit/ds/symlink/sn74hc245.pdf

If I go probing the circuit before I turn it on then start the panel it works fine sometimes.
If I leave it off for a while then turn it on, I get a green startup in some sectors and they stay green when the panel starts to do its thing.

Three MBI5051GP ICs do a sector of 8x8 leds.
When the panel runs fine, the SDI pin on one of the three MBI5051 ICs in that sector runs low.
When the panel doesn't run fine, the sectors that stay green, that same MBI5051 IC at the SDI pin runs high at 4.1V.
The other two MBI5051 ICs in the same sector have the same voltage at the SDI pin whether the panel is working or not.

I was thinking its a loose solder somewhere, but wouldn't know which one to check.
I was also thinking that maybe an octal bus transceiver is on the fritz and only works intermittently, not giving the right code to the MBI5051 ICs.
I'm stabbing in the dark a bit.

Would someone be able to shed some light on this please?

Thanks

Pete

[image]
[image]
[image]

BigPeteNZ, 2021-05-10:
Finally fixed it. A faulty MBI5051GP LED driver was too blame. It was preventing the other LED drivers from loading. Unsoldering the resistor in series with the voltage supply pin going to the MBI5051 chip, then comparing the resistance reading to other chips yielded chip with a different resistance. It was replaced and the board started operating normally.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin