hzeller/rpi-rgb-led-matrix #1222: Help me , Display not show

6010110254, 2021-01-01:
Help me
Several days ago I was able to show led-image-viewer, today I can't show it again, same code.
[image]

[image]

I don't know
Why
Thanks

davemaster, 2021-01-01:
... previous picture?

possible cause:
1. Power supply fault
2. data cable fault
3. rpi - rgb panel pcb of cable fault
4. command line is not the same as you believe, or you update hzeller library 

check them up

jaygarcia, 2021-01-07:
What Panels are you using? What is the connector type?

6010110254, 2021-01-08:
Sorry for saying late
Now I can show it, I don't know why.

closed 2021-01-08 by 6010110254