hzeller/rpi-rgb-led-matrix #1223: Adafruit bonnet & 32x64- only half the panel is working?

Tavros613, 2021-01-01:
I have a 32x64 panel and an adafruit bonnet, when I try any of the demos only half of the panel lights up (and it's every two rows). I'm not sure if I'm missing something here, but this has me stumped. 
[image]

hzeller, 2021-01-01:
Check out the corresponding topic in the README for this: https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet
You need to add some hardware mod to your Adafruit board.

closed 2021-01-01 by hzeller

davemaster, 2021-01-01:
try with --led-multiplexing= x, where x is a value according your RGB panels, in the help you can find that

Tavros613, 2021-01-05:
> Check out the corresponding topic in the README for this: https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet
> You need to add some hardware mod to your Adafruit board.

This isn't a 64x64 board is my confusion?

hzeller, 2021-01-05:
Oh I see, it is only 32 high. In that case what @davemaster said is something to check for: the led-multiplexing values, go through all of them to see if it makes a difference.

If you see that the visible rows somehow overlay (showing the things of the dark rows; not really visible in the chosen example above, try the rotating `-D0` square), then there might also be a problem with an address line; but since you have a premade Adafruit board, that might be somewhat unlikely. Could still be the ribbon-cable though.

Tavros613, 2021-01-05:
I have two LED boards and it happens the same on each with two different cables. My only remaining thought is it's either the PI or the bonnet?