hzeller/rpi-rgb-led-matrix #1224: led-matrix removed?

jax200, 2021-01-03:
Hi  Henner  I'm still playing with the rssparse.py code and in the original 2016 rendition it uses:

` os.system("sudo ./led-matrix -r 16 -c 3 -t 60 -m 25 -D 1 "+disp)`

yet "led-matrix" seems to be missing on the build.   Is this correct ?  Possible to get it back so I can get this to work?

Thanks,  Jack

hzeller, 2021-01-03:
The program moved to `examples-api-use/demo`

closed 2021-01-03 by hzeller

jax200, 2021-01-03:
Using demo for some reason it is not automatically changing feeds - need to use Ctrl-C.   I am going to try and rewrite for python3, but was hoping to first try and get the previous python 2 version working as in 2016.   I also just reinstalled via make and for some reason I didn't see the fonts directory?  /usr/share/fonts/  ... odd

hzeller, 2021-01-03:
there are fonts in the project, but they are not installed in `/usr/share/fonts`

The demo program is really only for playing and testing. If you want any kind of application, you should write it in C++ (or Python).

jax200, 2021-01-03:
I will do that..  Tried asking for help in discourse but no luck.