hzeller/rpi-rgb-led-matrix #1227: Panel HUB75E 128x64 and Raspberry Pi Zero W error, always big bars green and blue.

moybm, 2021-01-06:
Hello, first of all thank you for the possibility that I am given to share my problem.

I have a Rasberri Pi Zero W connected directly to a 64x128 panel and HUB-75E connector using the web schematic [https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md](url)  and when I run the demo I get a half green and half blue screen but no image (image capture). I want to try not to use any Adafruit HAT because I am only going to show only texts and make it as simple as possible to adapt it behind the screen

I have checked the wiring several times and everything is correct, following the pinout of :) 😄

I try too:
 _sudo ./demo -D 0 --led-rows=64 --led-cols=128 --led-gpio-mapping=regular-pi1_

and I don't know what else to review ... could you guide me or if someone knows what it could be?
Thanks a lot!

[image]

[image]

dpunzeng, 2021-01-24:
I think you need to leave away the "--led-gpio-mapping=regular-pi1" as it is for a Pi One and your pix shows a recent Pi Zero (it has  wifi on board)

moybm, 2021-01-25:
Hello friend, first of all thank you very much for having answered... after continuing testing I decided to reinstall everything completely, and re-download the library once downloaded I launch the command:

_make -C examples-api-use_

and it performs all the steps correctly without receiving an error, but when I execute the command now I get another different image but just as wrong:

_sudo ./demo -D0 --led-rows=64 --led-cols=128_

o

_sudo ./demo -D0 --led-rows=64 --led-cols=128 --led-row-addr-type=1_

and some more combinations of different parameters but still the same.
[image]

Many thanks 👍

closed 2025-11-27 by marcmerlin