hzeller/rpi-rgb-led-matrix #1228: Rpi 3a+ Adafruit Hat + Outdoor 40x40 ABC panel

OddmarDam, 2021-01-08:
Hi.

I have this panel:
[image]
[image]

There is a HUB75 connector, ABC addressing, 40x40 leds.

Only visible chips are these.
[image]
[image]

It’s part of a big outdoor screen.

When i run this demo:
sudo ./demo -D0 --led-brightness 20 --led-gpio-mapping=adafruit-hat-pwm --led-cols 40 --led-rows 20 --led-multiplexing 16 --led-chain 2

i get this result.
[image]

Only --led-multiplexing that shows any resemblance of a cube is 16 - P8Outdoor1R1G1

Can someone please help me. What is wrong here.

There are suggestions on making a new multiplexing mapper, but I do not know how.

Any help is appreciated.

hzeller, 2021-01-08:
if it is 40x40, does `--led-rows=40` create a better result ?

OddmarDam, 2021-01-08:
> 
> 
> if it is 40x40, does `--led-rows=40` create a better result ?

No, worse :)
And it also produces errors from the PixelMapper.
`Error in PixelMapper: (0, 20) -> (79, -1) [range: 80x20]`
[image]

hzeller, 2021-01-08:
ok, so it could be that the pixelmapper needs to be adapted; quite likely that the original author of that pixelmapper only had played with a 20x40.

You might need to write your own multiplex mapper. I suspect that the internal structure of this panel is organized as 80x20. So start with `--led-cols=80 --led-rows=20`. Then systematically go through these dots to understand what the underlying mapping is. E.g. use `examples-api-use/pixel-mover.cc` as a program to help you.

Then, create a new multiplex mapper in `lib/multiplex-mappers.cc` that reflects how this panel does things.

closed 2025-11-27 by marcmerlin