hzeller/rpi-rgb-led-matrix #1229: Parallel panels has gaps

DennisMamedov, 2021-01-11:
I didn't touch my rgb led panels for a while and I decided to come back to it. I have setup two 64x32 parallel (so 64x64) panels on my Raspberry Pi 2, however when running the demos the rotating square skips rows in between as seen [here](https://user-images.githubusercontent.com/29798218/104144516-c5f79f80-53c3-11eb-802a-ff08785f10fd.jpg).

There also seems to be a single row that has some leds light up slightly, and there is a vertical red line bottom left

I have tried every multiplexing option, reinstalled the pi and have checked the wiring carefully probably more than 5 times now. But with every demo and multiplexing option the exact same rows are skipped every time. 

Back when I first tried to realise my project the board worked flawlessly, but replugging it after ~4 months gives back these weird results

The command I'm running for the demo is:
`sudo ./demo --led-cols=64 --led-parallel=2 --led-no-hardware-pulse -D 0`

I'm running raspbian lite, and [this](https://www.tinytronics.nl/shop/en/lighting/64x32-rgb-led-matrix-256x128mm?limit=100) is the exact model I bought 2 of.

Any ideas?

hzeller, 2021-01-11:
It looks like your D-line is broken somewhere.  Rows with the upper bits are off, while the content of the lower bits are 'folded' over each other.

Given that you see that in the parallel panels, it means that it is somewhere on the board or the Pi and not an individual panel or cable to a panel. Check that there is no short on the adapter board. Could also be an issue in the Raspberry Pi if something happened to it (I once accidentally killed two GPIO pins by connecting 5V to it).

DennisMamedov, 2021-01-11:
> It looks like your D-line is broken somewhere. Rows with the upper bits are off, while the content of the lower bits are 'folded' over each other.
>
> Given that you see that in the parallel panels, it [...]

Looks like you're right. Tried to light up an individual LED with the same gpio pin as the D pin, and it did not work. 
Looks like I will have to wire everything up again tommorow to a different pi!

Thanks alot

closed 2021-01-11 by DennisMamedov

hzeller, 2021-01-11:
You might also use a different GPIO that is still working, manually wire it over to the old pin input on your adapter board and then change the [hardware-mapping.c](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c) file and recompile to use that instead for your D line. Currently, D is mapped to [`GPIO_BIT(25)`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c#L40) - but you can change that.