hzeller/rpi-rgb-led-matrix #1230: Using the following code, always get corrupted size vs. prev_size Aborted

notbrain, 2021-01-11:
```
$ sudo npm run example -- examples/dashboard.ts

> rpi-led-matrix@1.10.0 example /home/pi/src/rpi-led-matrix
> ts-node -P examples/tsconfig.json "examples/dashboard.ts"

corrupted size vs. prev_size
Aborted
```

Is what I always get in about 15 seconds after invoke. My simple code is below, am I doing something dumb that could prevent this? node v12.18.2, RPI4 with 2GB RAM. Running out of RAM? But is it something I can code around or a real bug in C? 

```
import {
  Font,
  LedMatrix,
  // FontInstance,
  // Color,
  LayoutUtils,
  HorizontalAlignment,
  VerticalAlignment,
} from '../src';
import { matrixOptions, runtimeOptions } from './_config';

const Colors = {
  Aquamarine: 0x7FFFD4,
  Black: 0x000000,
  Blue: 0x0000FF,
  Cyan: 0x00FFFF,
  Green: 0x00FF00,
  Magenta: 0xFF00FF,
  Purple: 0x800080,
  Red: 0x800000,
  White: 0xFFFFFF,
  Yellow: 0xFFFF00,
};

const wait = (t: number) => new Promise(ok => setTimeout(ok, t));

try {

  (async () => {

    const matrix = new LedMatrix(matrixOptions, runtimeOptions).afterSync(() => { });
    const font = new Font('tom-thumb', '/home/pi/src/rpi-led-matrix/fonts/tom-thumb.bdf');

    const alignmentH: HorizontalAlignment = HorizontalAlignment.Left;
    const alignmentV: VerticalAlignment = VerticalAlignment.Top;

    matrix
    .clear()
    .font(font)
    .fgColor(Colors.Red)

    const lines = LayoutUtils.textToLines(font, matrix.width(), 'Hello, matrix!');

    LayoutUtils.linesToMappedGlyphs(lines, font.height(), matrix.width(), matrix.height(), alignmentH, alignmentV).map(glyph => {
      matrix.drawText(glyph.char, glyph.x, glyph.y);
    });

    while(true) {
      matrix.sync();
      await wait(15000);
    }

  })();

} catch(error) {
  console.error(`${__filename} caught: `, error);

}
```

hzeller, 2021-01-11:
size vs. prev_size is not something that is ever compared or reported by this library. So it is probably somewhere in the node adaption, for which you probably should ask that project.

closed 2021-01-11 by notbrain

notbrain, 2021-01-11:
@hzeller kind of you to point that out so graciously, apologies.