hzeller/rpi-rgb-led-matrix #1232: Flickering Issues with Joy-It 64x64 Panel on Raspberry4

Typ260, 2021-01-13:
I got flickering issues on my Panel. Here is a Video of it (http://62.171.139.190/download/led_matrix_flicker.mp4). Even when lighting up just one LED in the Corner it flickers.
I used this instrustion manual to set up the matrix (https://files2.elv.com/public/25/2511/251186/Internet/251186_um_eng.pdf)

The Hardware:

- Raspberry 4
- Joy-It 64x64 LED Matrix
- High Quality 230 V AC to 5 V DC power adapter

Things i tried:

- played with every possible command line argument, whats seen on the video is ` sudo examples-api-use/demo -D 0 --led-rows=64 --led-cols=64 --led-slowdown-gpio=3 --led-show-refresh --led-no-hardware-pulse` which is the best i can get. With more or less GPIO slowdown it gets worse.
- ordered a more expensive, higher rated power adapter, same issue
- rewired the matrix to the RPi 3 times, none of the times made it better
- tried every demo and util program

I appreciate every support i can get in this matter.

hzeller, 2021-01-13:
Here might be possibly some hardware factors involved

  * Are you using a line level adapter, like the Joy-It people suggest, or any other [active adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter) from this project ? Wiring up the Pi directly is not necessarily a recommended way: the 3.3V voltage level is not compatible with the usually 5V voltage level needed by the panels. And driving long cables with the GPIO outputs are also not a good idea.
  * Keep the cables _short_. From looking at your video, you have a fairly long flat cable which is problematic with the high frequencies involved. In particular if there is no buffer in-between (the voltage level adapter from above).

For  the software side: In general, some faint flicker can happen as the Raspberry Pi is not necessarily meant to output data in high-accuracy timing. But there are things you should minimize problems, such as using DietPi as operating system and switching of unnecessary things ( The [troubleshooting section in the README](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting) covers these. Also make sure to [set the `isolcpus` value](https://github.com/hzeller/rpi-rgb-led-matrix#cpu-use) ).

These things can also be related to slightly different timing requirements on LED panels, e.g. at which time they like the row to be switched and latched. But that is specific to your panel and might or might not be an issue here. But in particular these kind of glitches along the edges might be.
(Given that you bought a product from some company that uses my library, maybe you should ask them for support. This project is not affiliated with them in any way)

Typ260, 2021-01-13:
Thank you very much for answering.

Ill look into shortening the cables since i tried everything else.
Im using a line level adapter. Im also ordering the Joy-It Controllerboard, just to make sure.

Typ260, 2021-01-15:
**Update:**

I tried shorter cables which did nothing for my problem. I also ordered the controller board that Joy-It offers and the matrix now works fine.

So we learn: _Even though Joy-It says that their 64x64 matrix works when connecting directly to the GPIO, it **doesnt**. It works only with the additional controller board. - Tested with the Raspberry 4_

As far as i am concerned, this issue is resolved now.